nightscout/AndroidAPS #3675: Getting Error Unsuccessful with NSClientV3

jvmaanen, 2025-01-02:
NSClientV1 is syncing without issues. I switched to NSClientV3 because after switching to AAPS 3.3 I had some issues with the database and cleaned it. I want to import TDD information from my NS so that I can use Dynamic ISF without having to wait for 7 days.
I did a full sync during the whole night, but no TDD data was imported from NS.

Even with all sync options switched off in the NSClientV3 I get the error message. I also started a discussion on this topic on Facebook on January 1st, but didn't get a solution yet.

[image]

MilosKozak, 2025-01-02:
this is error in communication with NS, not resolvable by AAPS
you can restart phone, NS, try different internet connectivity ....

closed 2025-01-02 by MilosKozak

jvmaanen, 2025-01-02:
I restarted my phone, and I can access NS on my phone without a problem when on NSClientV1. But I noticed that when I'm on NSClientV3 I cannot access NS on my phone or computer. When I pause NSClientV3 I can access NS again on my phone and computer. My NS is hosted on Azure.