nightscout/AndroidAPS #3679: WatchFace (APPS Large) not right rendered after upgrade from AAPS 3.2.0.4 to AAPS 3.3.0

frice79, 2025-01-02:
After upgrade from 3.2.0.4 to 3.3.0 the Custom WatchFace AAPS(Large) is not properly rendered.

In particulary the white row ("MatchDivider")  in the middle of the screen is not visibile, and also the value inside the MatchDivider.

Thanks to @Philoul I receive a workaroud to  show the matchdiveder again:

Workaround is to switch on then switch off each parameter of AAPS(Large) to get correct visibility of watchface...
So for the background and white bar:
Set Dark parameter enabled, then disabled (or opposite)
Set matchdivider to enabled, then disable (or opposite)...
This should let you have again the divider bar visible...

In case of overlapping is necessary to change manually the value of: "min ago", "Uploader Battery" and "Status information"

Philoul, 2025-01-02:
This issue is not dedicated to AAPS(Large) Custom watchface, but due to a fresh install of Wear apk, then the default values shown within watch for each parameter, and the value read within CWF can be inconsistant.

I will have to deep dive within preference management to fix it 😉 (Hope it will be limited to Custom Watchface only, but currently not 100% sure)..

Thanks for this catch 👍

Philoul, 2025-01-02:
After a quick review of Custom WF code, I found (I think) the root cause...
The good news is it should be limited to CWF...
Will be able to provide a fix soon 😉.

Philoul, 2025-01-03:
Fixed by 79edafef19110b7dfc00f39da7663303142140e1

PR #3695 (2025-01-04): CWF Fix Default Preference management
Fix #3679
closed 2025-01-04 by MilosKozak