nightscout/AndroidAPS #3688: [3.3] Faulty calculation of delta

quizzmaster, 2025-01-03:
With the current master (3.3) I face a serious problem with the calculation of the deltas. Most of the time, the delta is calculated as zero although the bg values show differences:
[image]
[image]

The graph also shows many pale bg values:
[image]

I can provide more informations or logs if needed

olorinmaia, 2025-01-03:
I use G7 with xDrip direct connection 3.3 master and Fake data source 3.3 dev and I don't experience this.

Can you provide more info.
Sensor type, setup in regards to getting bg to AAPS, logs from AAPS?

quizzmaster, 2025-01-03:
I use the Libre 3 via Juggluco -> xDrip (Nightscout Follow) -> AAPS.

Here is an excerpt from the log (cut at the invokation of onReceive of the Data Receiver):
```
12:57:56.184 [main] D/CORE: [DataReceiver.onReceive():38]: onReceive com.eveningoutpost.dexdrip.BgEstimate Bundle{ com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200; com.eveningoutpost.dexdrip.Extras.BgSlope => 3.3557046979865773E-6; com.eveningoutpost.dexdrip.Extras.VersionInfo => 2124123110 8ed51e7-2024.12.31; com.eveningoutpost.dexdrip.Extras.Raw => 0.0; com.eveningoutpost.dexdrip.Extras.Display.Units => mg/dl; com.eveningoutpost.dexdrip.Extras.Time => 1735905473000; com.eveningoutpost.dexdrip.Extras.CalibrationInfo => -1; com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => 1; com.eveningoutpost.dexdrip.Extras.SensorStartedAt => 1735856133660; com.eveningoutpost.dexdrip.Extras.Noise => 2.282510757446289; com.eveningoutpost.dexdrip.Extras.SensorBattery => -1; com.eveningoutpost.dexdrip.Extras.NoiseWarning => 0; com.eveningoutpost.dexdrip.Extras.BgSlopeName => Flat; com.eveningoutpost.dexdrip.Extras.BgEstimate => 110.0; com.eveningoutpost.dexdrip.Extras.SourceDesc => Nightscout; com.eveningoutpost.dexdrip.Extras.SourceInfo => Nightscout Follow; }Bundle
12:57:56.267 [DefaultDispatcher-worker-6] D/BGSOURCE: [XdripSourcePlugin$XdripSourceWorker.doWorkAndLog():106]: Received xDrip data: Bundle[{com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel=200, com.eveningoutpost.dexdrip.Extras.BgSlope=3.3557046979865773E-6, com.eveningoutpost.dexdrip.Extras.VersionInfo=2124123110 8ed51e7-2024.12.31, com.eveningoutpost.dexdrip.Extras.Raw=0.0, com.eveningoutpost.dexdrip.Extras.Display.Units=mg/dl, com.eveningoutpost.dexdrip.Extras.Time=1735905473000, com.eveningoutpost.dexdrip.Extras.CalibrationInfo=-1, com.eveningoutpost.dexdrip.Extras.NsNoiseLevel=1, com.eveningoutpost.dexdrip.Extras.SensorStartedAt=1735856133660, com.eveningoutpost.dexdrip.Extras.Noise=2.282510757446289, com.eveningoutpost.dexdrip.Extras.SensorBattery=-1, com.eveningoutpost.dexdrip.Extras.NoiseWarning=0, com.eveningoutpost.dexdrip.Extras.BgSlopeName=Flat, com.eveningoutpost.dexdrip.Extras.BgEstimate=110.0, com.eveningoutpost.dexdrip.Extras.SourceDesc=Nightscout, com.eveningoutpost.dexdrip.Extras.SourceInfo=Nightscout Follow}]
12:57:56.286 [RxCachedThreadScheduler-156] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():62]: Firing EventNewBG GlucoseValue(id=32222, version=0, dateCreated=1735905476272, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1735905473000, utcOffset=3600000, raw=0.0, value=110.0, trendArrow=FLAT, noise=null, sourceSensor=LIBRE_3_FOLLOW)
12:57:56.289 [RxCachedThreadScheduler-156] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():67]: Firing EventNewHistoryData 1735905473000 GlucoseValue(id=32222, version=0, dateCreated=1735905476272, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1735905473000, utcOffset=3600000, raw=0.0, value=110.0, trendArrow=FLAT, noise=null, sourceSensor=LIBRE_3_FOLLOW)
12:57:56.292 [RxCachedThreadScheduler-156] D/DATABASE: [PersistenceLayerImpl$insertCgmSourceData$4.apply():586]: Inserted GlucoseValue from Xdrip GlucoseValue(id=32222, version=0, dateCreated=1735905476272, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1735905473000, utcOffset=3600000, raw=0.0, value=110.0, trendArrow=FLAT, noise=null, sourceSensor=LIBRE_3_FOLLOW)
12:57:56.296 [DefaultDispatcher-worker-6] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.plugins.source.XdripSourcePlugin$XdripSourceWorker Data {}
12:57:56.346 [DefaultDispatcher-worker-1] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.plugins.sync.nsclientV3.workers.DataSyncWorker Data {}
12:58:01.309 [pool-10-thread-1] D/WORKER: [CalculationWorkflowImpl.stopCalculation():47]: Stopping calculation thread: onEventNewHistoryData
12:58:01.334 [pool-10-thread-1] D/WORKER: [CalculationWorkflowImpl.stopCalculation():52]: Calculation thread stopped: onEventNewHistoryData
12:58:01.342 [pool-10-thread-1] D/WORKER: [CalculationWorkflowImpl.runCalculation():64]: Starting calculation worker: EventNewBG to 03.01.25 12:58:01
12:58:01.510 [DefaultDispatcher-worker-2] D/CORE: [AutosensDataStoreObject.createBucketedDataRecalculated():239]: Adjusted time 03.01.25 12:57:53
12:58:01.544 [DefaultDispatcher-worker-2] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.workflow.LoadBgDataWorker Data {}
12:58:01.582 [DefaultDispatcher-worker-2] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.workflow.PrepareBucketedDataWorker Data {}
12:58:01.616 [DefaultDispatcher-worker-2] D/WORKER: [LoggingWorker$doWork$2.invokeSuspend():27]: Worker result SUCCESS for class app.aaps.workflow.PrepareBgDataWorker Data {}
[... cut ...]  fit a0: 0,00 mg/dl Parabola fit a1: 0,00 mg/dl/5m Parabola fit a2: 0,00 mg/dl/(5m)^2
13:03:04.516 [RxCachedThreadScheduler-51] D/GLUCOSE: [GlucoseStatusProviderImpl.getGlucoseStatusData():264]: Glucose: 124 mg/dl Noise: 0 Delta: 0 mg/dlShort avg. delta:  0,00 mg/dl Long avg. delta: 3,62 mg/dl Dura ISF minutes: 0,00 m Dura ISF average: 124,00 mg/dl Parabola minutes: 0,00 m Parabola correlation: 0,00 Parabola fit a0: 0,00 mg/dl Parabola fit a1: 0,00 mg/dl/5m Parabola fit a2: 0,00 mg/dl/(5m)^2
```

olorinmaia, 2025-01-08:
It happend to our setup but only for 1 reading. Seems like delta is not calculated when a new sensor is started with different time interval on BG arrival than previous sensor. I sent Milos screens and logs on discord.

After a few readings delta is calculated and again and hasn't occurred again during this sensor session. 

Setup is G7 and direct connection via xDrip+

sanderrenes, 2025-01-10:
I have the same issue with 3.3.0.
Using libre2, Xdrip+ and  oop2

The delta is not always 0, but it is wrong. The 5 minutes delta and 15 minutes delta appear to be the same always and the 40  minute delta is some other number. The strangest thing is that the arrow does seem to agree with Xdrip measurements.

I tried installing Juggluco next to Xdrip to see if that helped. It did not. 

since AAPS started intervening based on the incorrect deltas, I had to downgrade again.

painbrain81, 2025-01-11:
same problem here (pic is from client but master is the same)
the previuos value is 7 minutes ago. maybe the calculation of deltas is bugged when is out of "standard" 5 minutes?

config: 3.3.1.0 + xdrip + oop2 (libre2+omnipod)
[image]

MilosKozak, 2025-01-13:
someone affected: give me credentials to setup AAPSClient with the bug

bamloeff, 2025-01-19:
Warning: due to this issue Automations that make use of delta's as conditions will not work as expected.

bamloeff, 2025-01-19:
@MilosKozak I'm experiencing this issue too. I'm currently on AAPS v3.3.1.0, FSL2 with Xdrip/OOP2, Dana-i and Nightscout v3. What exactly do you need to set up your test?

BIO-DISASTER, 2025-01-19:
I'm also have this issue, with AAPS 3.3.1.0 - FSL2 - Xdrip - Juggluco or OOP2. One thing I noticed is that if I scan FSL2 with NFC in Xdrip, Delta is doing something.

bamloeff, 2025-01-26:
Could it be this has something to do with the recalculations AAPS frequently does on FSL2 values? Can't really support this with data, but I noticed there are a lot of open/transparent dots in my BG since version 3.3. I suppose this point at recalculated BG's, can't find that in the docs though. I suspect that when such a recalculated BG value occurs the delta's may not be calculated correctly.

BIO-DISASTER, 2025-01-26:
> I'm also have this issue, with AAPS 3.3.1.0 - FSL2 - Xdrip - Juggluco or OOP2. One thing I noticed is that if I scan FSL2 with NFC in Xdrip, Delta is doing something.

I would like to add that I left Xdrip out and sticked with only FSL2 - Juggluco - AAPS and the delta values are coming through just as you would expect. So I still think the issue is somehow Xdrip related and not so much linked to the FSL2 specific

sanderrenes, 2025-01-27:
> I would like to add that I left Xdrip out and sticked with only FSL2 - Juggluco - AAPS and the delta values are coming through just as you would expect. So I still think the issue is somehow Xdrip related and not so much linked to the FSL2 specific

Wait, if you run juggluco without xdrip, you get deltas. Are they the correct deltas?

MilosKozak, 2025-01-27:
repeating: if you can replicate it in AAPSClient I can have a look

sanderrenes, 2025-01-27:
> repeating: if you can replicate it in AAPSClient I can have a look

I send you Nightscout details via Facebook DM, did that work?

I'm currently on 3.3.1.0, when I send you the link I was on 3.3.1.3. Both have the same error

MilosKozak, 2025-01-27:
i need admin token to be able connect vie AAPSClient
try it first yourself to be sure the issue is replicable there

sanderrenes, 2025-01-28:
> i need admin token to be able connect via AAPSClient try it first yourself to be sure the issue is replicable there

It replicates in AAPSClient. 
I can send you the NightScout details, where do you want them?

sanderrenes, 2025-02-02:
> Wait, if you run juggluco without xdrip, you get deltas. Are they the correct deltas?

I tried this with my new sensor (fsl2). I removed xdrip from phone, started with just juggluco and aaps (3.3.1.0). This seems to give correct deltas.
Aaps warns it uses recalculated glucose values (yellow / orange triangle top left of glucose value), but that is normal for fsl2

If you try this, don't forget to tell aaps to upload your glucose to NS instead of xdrip

Current setup fsl2 (eu) - - > juggluco - - > aaps 3.3.1.0

bamloeff, 2025-02-02:
> I tried this with my new sensor (fsl2). I removed xdrip from phone, started with just juggluco and aaps (3.3.1.0). This seems to give correct deltas. Aaps warns it uses recalculated glucose values [...]

But if you let FSL2 values come in directly through Juggluco they will be updated every minute, isn't it? What will this do to the closed loop?

sanderrenes, 2025-02-02:
> > > > 

SHORTENED FOR CLARITY

> 

> But if you let FSL2 values come in directly through Juggluco they will be updated every minute, isn't it? What will this do to the closed loop? There also a second route: sending FSL2 values from Juggluco to Xdrip and then to AAPS. This will make values come in every 5 minutes and will enable smoothing in Xdrip. But will it also solve the delta's problem?

I tried to get them send through xdrip, did not get it to work properly. I have not used juggluco before, so might have messed that up. 

Right now it updates every minute. The deltas are recalculated to cover the right period, so they are in the range of values you would expect. 

My loop seems to respond normally, in the sense that i did not see much of a change relative to just before deinstalling xdrip. 
It might be responding more often, but not fundamentally different by the looks of it.

Jan1758, 2025-04-05:
Configuration
AAPS 3.3.2.0 
FSL2
DIABOX
NO APP XDRIP

I have the same problem. The delta values ​​do not match the table of glucose values. 

**I also notice that the graph of glucose (sometimes?) is based on the incorrect delta values. See the attached screenshots.**

[image]
[image]
[image]
[image]

I am concerned that other calculations also use the incorrect delta values. For example, the required carbohydrates calculated by AAPS, in the event of an approaching hypo, are (sometimes?) unrealistically high. Unfortunately, I do not have any screenshots of that.

Please, let me know if and how I can help resolve this issue.

Jan van den Berg

MilosKozak, 2025-04-05:
Provide logs if you need help

sanderrenes, 2025-04-05:
I can give you nightscout credentials, where do you want them?

I also just opened another issue about the calculations of the deltas.
There are some odd things in the math.
If you or anyone is working on them, please check those out too.

Jan1758, 2025-04-06:
[AndroidAPS_LOG_1743920220258_log.zip](https://github.com/user-attachments/files/19620212/AndroidAPS_LOG_1743920220258_log.zip)

A first log. Usable?

Jan

sanderrenes, 2025-04-06:
> [@sanderrenes](https://github.com/sanderrenes) [@Jan1758](https://github.com/Jan1758) sorry, I need screenshots and logs from the same time

Here you go:

https://filesender.surf.nl/?s=download&token=3ae368e9-89ea-4e8c-a8e6-a06aa8349e64

three screenshots over the last 10 minutes and the live log file as of now

MilosKozak, 2025-04-06:
@sanderrenes you are using 3 min interval which makes everything way more complicated (except it's not recommended and contraproductive)
setup 5 min and test again

Jan1758, 2025-04-06:
I changed the setup in Diabox in 5 minuten. Half an hour later I've made the sreenshots and logfile within 1 minute. 

[image]
[image]

[DOC-20250406-WA0003.zip](https://github.com/user-attachments/files/19622586/DOC-20250406-WA0003.zip)

MilosKozak, 2025-04-08:
did you wait at least 36h?

sanderrenes, 2025-04-08:
@Jan1758 said half an hour later, so i guess not.

I've reinstalled xdrip, checked settings etc. Now on 6 minute interval, so far same picture as Jan. Short Deltas are both zero. 

Ill wait till the 36h are up and make logs

Jan1758, 2025-04-08:
@MilosKozak @sanderrenes No, this morning I see the effects in Diabox. Measurements with an interval > 5 minutes. I see no effects in AAPS. I will send new logs and screenshots today.

Jan1758, 2025-04-08:
@MilosKozak @sanderrenes I see that the Arrows do not match my expectations. See screenshot. Between the upper 2 measurements is a delta of 2.0. The arrow is single. I suppose it should be double. 
I see more "wrong" arrows in this screenshot.  (In the docs I could not find the exact zones of the arrows. Sorry.)

Later on I will send more.

[image]

. The arrow is single but should be double.

Jan1758, 2025-04-08:
New screenshots and logs. 
1 delta's do not match with the table
2 graph nearby horizontal where I expect an arc because of measurements 
3 some arrows in the table are wrong. 

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Every screenshot of a graph are connected to 1 or 2 screenshots taken on the same moment.

MilosKozak, 2025-04-08:
well 6 min is even worse ..... can you set it to 5 min?

MilosKozak, 2025-04-08:
displayed arrrows are exactly how received from xdrip
```
08:31:25.365 [RxCachedThreadScheduler-2422] D/DATABASE: [PersistenceLayerImpl$insertCgmSourceData$4.apply():587]: Inserted GlucoseValue from Xdrip GlucoseValue(id=26488, version=0, dateCreated=1744093885353, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=null, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1744093877047, utcOffset=7200000, raw=0.0, value=171.0, trendArrow=SINGLE_UP, noise=null, sourceSensor=UNKNOWN)
```

Jan1758, 2025-04-09:
> EDITS: added figure of table instead of unreadable HTML

@sanderrenes This is adressed to Milos?

The arrows are not the main problem. I do not use Xdrip+. In AAPS a plugin named Xdrip imports the data from Diabox. Today I will compare the data from Diabox and AAPS. 

Let's focus on the faulty calculations

Jan1758, 2025-04-09:
@MilosKozak @sanderrenes 
> well 6 min is even worse ..... can you set it to 5 min?

Diabox (with FSL2) has two possibilities: 5 minutes  on or off . Finetuning is not possible. 
5 minutes off: short deltas not always zero.
5 minutes on: short deltas always zero.

[image]

MilosKozak, 2025-04-09:
set 5 min, wait 2 days, then provide screenshots and logs

Jan1758, 2025-04-09:
@MilosKozak 
In my contribution of yesterday 8th Diabox was set to 5 minutes two days before. Meets your requirements. 
You want me to send more/new/other info? 
You have located the problem?

Jan1758, 2025-04-09:
Not possible to finetune Diabox. It is 5 minutes on or off.

bamloeff, 2025-04-09:
That's what I see also when using Xdrip direct mode (OOP2) with FSL2: updates every 6 minutes. Can't be changed in Xdrip. Looks like AAPS is doing a lot of recalculations (transparent dots in the main graph).

[image]

sanderrenes, 2025-04-09:
There appears to be a 2-minute interval for the data collection in Juggluco in my case. If Juggluco only gets 2-minute data, it would be impossible to get a 5-minute interval without recalculations. It would be 2, 4, or 6 minutes, and all of our apps seem to go for the 6-minute version. 
Neither Juggluco, nor Xdrip allows me to set an interval in any other interval. 

What I don't get is that the FSL2 sensor, according to the producers, captures data every minute and submits this to the app. 
https://www.abbott.com/corpnewsroom/diabetes-care/how-sensors-and-insulin-pumps-connect-diabetes-care.html

Jan1758, 2025-04-09:
I notice that the arrows in Diabox and in AAPS are not always the same. Diabox calculates a 2-minutes interval. Sometimes the 2-minutes in Diabox differ from a 5- or 6-minutes interval in AAPS. I only use Diabox and AAPS. I agree with @sanderrenes and @bamloeff that there are a lot of recalculations in AAPS. 

I noticed also that the 2 minutes interval results in several deltas not zero on the screen and the "5"-minutes interval only zeros. 

What are the consequences in AAPS?

Jan1758, 2025-04-09:
> [@Jan1758](https://github.com/Jan1758) EDITS: added figure of table instead of unreadable HTML
> 
> > " (In the docs I could not find the exact zones of the arrows. Sorry.)"
> 
> I found them in the code while I was looking for the delta calculations in the file TrendCalculatorImpl.kt:
> 
> ```
>        return when {
>             slopeByMinute <= -3.5 -> TrendArrow.DOUBLE_DOWN
>             slopeByMinute <= -2   -> TrendArrow.SINGLE_DOWN
>             slopeByMinute <= -1   -> TrendArrow.FORTY_FIVE_DOWN
>             slopeByMinute <= 1    -> TrendArrow.FLAT
>             slopeByMinute <= 2    -> TrendArrow.FORTY_FIVE_UP
>             slopeByMinute <= 3.5  -> TrendArrow.SINGLE_UP
>             slopeByMinute <= 40   -> TrendArrow.DOUBLE_UP
>             else                  -> TrendArrow.NONE
>         }
> ```
> 
> This is per minute, the delta we usaly see is per 5 minutes. Also, I think this is the US measures (mg/dl) not mmol/l that we use in Europe (divide by 18 multiply by 5 to translate), so the lower and upper bounds, in 5 minute average change:
> 
> [image]

I suppose "40" has to be "4.0"?

Jan1758, 2025-04-10:
@MilosKozak @sanderrenes @bamloeff 
This morning I foresaw I'm going low. But AAPS did not give me an advise. Why? I suppose because the deltas are zero and the graph is horizontal. 

[image]
[image]
[image]

Next measurement. Surprise. A caculated delta! -0,8. 

[image]
[image]

The faulty calculaties also effects crucial funtions in AAPS. 

I will add the logs later.

Jan1758, 2025-04-10:
@bamloeff 
> As I told you a couple of times already as far as I can see our loops are working as they should.

Bart, 

Thanks for your concern, but that is not necessary. I know from experience that quality benefits from good feedback. As long as the cause of this problem has not been found, I will report everything that I notice that could be related to this. I do not want to suffer from tunnelvision because everything(?) would work fine. Be more (constructively) critical. Moreover, we are responsible for the app we use. That is why I want to be able to make my contribution if I see that something in the app is not going well. 

v.g. Jan

sanderrenes, 2025-04-10:
Jan, Bart en Milos,

Goed nieuws,
There is progress

[image]

I'll push these changes to my branch on github, so you can see what i did. It's in one line of code, an if statement. 

Ill fill out the details from the pc

Met vriendelijke groet / regards
Sander

sanderrenes, 2025-04-10:
The details: 

https://github.com/sanderrenes/AndroidAPS/commit/dfbe47925c8c9d345d884f488c70e24e8c4efb2a

I changed the if statements on line 71  to exclude the filtering on recalculation. Not sure why it is recalculated data, but I suspect the smoothing that is done on fsl2 data, or the difference in measurement interval is causing the issue in one-or-the-other of our apps. 

With this change, I have better-looking deltas on my FSL2 with juggluco. I'll do some more spot-checks this weekend. 

odd observation:
If you go down through the code in the GlucoseStausProvider class, you still see shifts between the two different Glucose measurement columns, e.g.:

71:   if (data[i].recalculated > 39)
117: if (data[i].value > 39 )

I am not sure this is intentional.

Jan1758, 2025-05-22:
@MilosKozak  Has there been any progress on this issue? Has @sanderrenes perhaps found the solution or do you still need more info?

sanderrenes, 2025-05-23:
@Jan1758 thanks for the reminder, I wanted to check some things.
The statement that is causing problems was introduced with version 3.3.0

in 3.2.0.4 the relevant bit of code is:
if (data[i].recalculated > 38) {
...
change = now.recalculated - then.recalculated
val avgDel = change / minutesAgo * 5

and in 3.3.0 this becomes:
if (data[i].value > 39 && !data[i].filledGap) {   // if (data[i].recalculated > 38) {
....
change = now.recalculated - then.recalculated
                val avgDel = change / minutesAgo * 5

The release notes did not make clear to me why the extra condition was added (or why the check move to value instead of recalculated)

I then tried to find the filledGap variable, but got a bit stuck. It appears to be set deep in the code, I can follow it back till "inMemoryGlucoseValue.kt" where I also found this comment:

/**
 * Simplified [app.aaps.core.data.model.GV] for storing in memory and calculations
 * It may correspond to GlucoseValue value in db
 * but because of 5 min recalculations and smoothing it may not
 */

That seems to be our issue. Since my FSL2 reports on a 2-minute interval, there is never a measure exactly 5 minutes ago. To get a 5-minute measure, you have to fill the gap between 4 and 6 minutes by definition. 
This 2 minute interval appears particular to this sensor. 

So, is this what causes your error? if you have FSL2 - EU version like me, then very likely.
Can you solve it?
yes, either make the same adjustment as I did in your own fork, or use my fork (i am still running on it, works fine with me)

Can we solve it for everyone? 
That is harder. It seems it is particular to one sensor. I don't know why this if-condition was added, but if it is for safety reasons, it might not be wise to bluntly remove it for everyone. We could work around it, for instance with an if condition that says something along the lines:

if (data[i].value > 39 && (!data[i].filledGap || SourceSensor=="Libre2") 

but that makes the code a lot harder to read, and I'm not sure it is all that clear why this would be necessary to have. So for maintainability of the code that is not nice. 
This last part can be fixed by adding comments like the one about the recalculations, if that is ok with the core devs like @MilosKozak

Jan1758, 2025-06-01:
@sanderrenes

I've done the following by now. 
First, I replaced Diabox with Juggluco. That went smoothly. I notice that with Juggluco the Bluetooth connection is more stable. There are almost no gaps in the BG data series. 
A few days later I replaced FSL2 with One+. This also went smoothly. One of the differences between FSL2 and One+ is the interval. In theory, FSL2 has a measurement every minute and One+ every 5 minutes. Furthermore, it is striking that a continuous and streamlined curve of BG's is now emerging in AAPS. And AAPS is more stable. No more "incomprehensible" actions in the form of "unwanted" boluses. 

Back to the issue. The attached screenshot shows that the 3 lines with the deltas of BGs differ. The problem with One+ does not occur. For the time being I will continue with the combination One+, Juggluco and AAPS. 

This will be my last contribution to this issue. 

[image]

MilosKozak, 2026-10-11:
Since 3.4.0 the delta also uses gap-filled (recalculated) glucose values. Skipping them was the cause of delta 0 with Libre 2/3 data. Please open a new issue with logs if you still see wrong deltas on a current version.

closed 2026-10-11 by MilosKozak