nightscout/AndroidAPS #3692: Config Builder Samsung Tizen

reijoahola-nightscout, 2025-01-03:
This setting enables data broadcast to several CGM apps serving several types of watches, not only Samsung Tizen.
Known cases are
- G-Watch Wear App (Tizen watches)
- G-Watch Wear App (all Wear OS 3 and 4 watches, not only Samsung Wear OS)
- GlucoDataHandler app (all Wear OS 3, 4 and 5 watches)
- GlucoDataAuto (AndroidAuto companion app)

I leave it for further discussion what the header and description text should be in this setting.

olorinmaia, 2025-01-03:
You could try asking AI for suggestions to this 😉 I agree we should find a better name and description

olorinmaia, 2025-01-03:
Suggested Setting Name:
"Companion App Broadcast"

Description:
Enables data broadcasting to companion apps on Samsung Tizen, Wear OS, and Android Auto devices, including G-Watch, G-Watch Wear, GlucoDataHandler, and GlucoDataAuto.

olorinmaia, 2025-01-03:
To illustrate:

| Current string | Improved string |
|--------|--------|
| [image] | [image]|

reijoahola-nightscout, 2025-01-04:
It is OK for me :)

MilosKozak, 2025-01-04:
what is added value on android (compared to native AAPS possibilities)?

olorinmaia, 2025-01-04:
I didn't know about several of these apps (because of old description, I thought this only was for Tizen OS). I've tested them and they add nothing new entirely except for perhaps G-Watch that allows for showing BG graph + complications. Glucodatahandler have a few more options in complications than in-built with AAPS.
[image]

More info:

https://play.google.com/store/apps/details?id=sk.trupici.gwatch.wear&hl=en
https://github.com/pachi81/GlucoDataHandler/blob/master/README.md
https://github.com/pachi81/GlucoDataAuto

reijoahola-nightscout, 2025-01-04:
> what is added value on android (compared to native AAPS possibilities)?

G-Watch Wear App is just another alternative for analog and digital watchface having CGM graph.

GlucoDataHandler provides several types of complication values for other watchfaces. This has been usefullespecially in Galaxy Watch Ultra and Galaxy Watch 7 that has Wear OS 5 directly from the factory and AAPS watcfaces are very difficult to use (switching between AAPS and other watchfaces).

GlucoDataAuto has a better support for showing BG value in AndroidAuto.

All these companion apps receives BG/CGM data very reliably from AAPS app (compared to data reading from cloud services).

olorinmaia, 2025-01-04:
@reijoahola-nightscout BTW. Wear communications have been improved in 3.3, so you can use other watchfaces with AAPS complications and the tiles should now work.

reijoahola-nightscout, 2025-01-04:
I think G-Watch app developer took this data broadcasting interface into use at first when providing watchface for Galaxy Tizen watches. Then he continued to use the same inteface when G-Watch Wear App was developed later (Wear OS watches).
Then GlucoDataHandlerApp developer copied the same method when he designed own apps (GDH, GDA).

So actually it is not a question about some specific watch type and operating system but just a phone application that can do all kind of addional services.

reijoahola-nightscout, 2025-01-04:
> @reijoahola-nightscout BTW. Wear communications have been improved in 3.3, so you can use other watchfaces with AAPS complications and the tiles should now work.

Yes, I know but watchfaces are very complicated to use because of factory installed Wear OS 5. I have also Galaxy Watch 4 that is now upgraded into Wear OS 5 and it has no problems with watchfaces designed earlier (xDrip+, AAPS, G-Watch Wear).

reijoahola-nightscout, 2025-01-04:
BTW, G-Watch Wear, GDH and GDA can receive data also from xDrip+ but it is not so easy as with AAPS and AAPS alternative works much reliably. With xDrip+ Inter-app settings seems to be often a problem and "Identify Receiver" is needed.

MilosKozak, 2025-01-04:
Hearing this I'd keep the naming as it is (announce for Tizen only). Most of the features AAPS supports natively which is preffered way

closed 2026-09-23 by olorinmaia