nightscout/AndroidAPS #3694: Gap between Automation Buttons too big on small phone

Sneepie, 2025-01-03:
The automation buttons are (too) big on a small phone (e.g. Cubot KingKong Mini3)  
Please see pics below, first the old/normal view second the todays updated/new view.

Suggestion: extra space via user input-text and not hard coded.

[image]
[image]

Philoul, 2025-01-04:
In my mind the little space between each button is better compare to the overall grey bar for 5 different buttons we had before.
We should probably reduce the padding within each button to give more place for the text inside (in your screenshot about 2/3 of the width of each button is lost for the text and is in grey...

Sneepie, 2025-01-04:
I completely agree. Even with cropped text (please see pic) there still is (imo too much) lost space.

[image]

ga-zelle, 2025-01-04:
We can also think about the logo. It ios the sam for each button and therefroe does not add any value. Either use the logo for the (first?) action triggered or leave it out completely.

MilosKozak, 2025-01-04:
improved here https://github.com/nightscout/AndroidAPS/commit/3751ef73ce017691aa434a46e53175c194dd3688

olorinmaia, 2025-01-04:
@MilosKozak It looks good. NIce with icon for each Action. It seems like the Profile action icon and start Temp target action icon isn't working. Could an icon be added to them also, else show the old icon?

[image]
[image]

olorinmaia, 2025-01-04:
@MilosKozak looking good with 56aaf854
[image]

closed 2025-01-04 by MilosKozak

ga-zelle, 2025-01-04:
Sorry, why is the color lost? TEST2 looks like an OK one although it is a STOP with red quare in the original icon.

MilosKozak, 2025-01-04:
on the line bellow the text match the color icon

olorinmaia, 2025-01-04:
I think it's cleaner with same color on icon/text, but either is fine.

Sneepie, 2025-01-04:
more logical and size adjustable now. way better. thanks
[image]