nightscout/AndroidAPS #3698: [Request] Ad name of patient to homescreen and widget for AAPS Client 1 and 2

gerison77, 2025-01-04:
With the update to 3.3.0.0. came a dual watchface which showed the name of the patients to distinguish the data.

The same problem and the same solution would apply to the widgets of AAPS Client 1 and 2
as well the applications itself. If you start the app by clicking on the watchface, you have no hint on the homescreen to know whose data are shown.

I had the workaround to put the name of the patient in the profilename, which is shown on the homesreen. You can see the profilename in the widget as well.

Since I use autotune in DEV with automations, the profilename can't be altered. If just shows "adjusted" or "Angepasst" in german.

Would be nice to see this feature in future versions

MilosKozak, 2025-01-04:
open for discussion
profile name is from my point better solution than wasting of space

gerison77, 2025-01-04:
@MilosKozak I agree, that it would suffice. That would require to alter the name of the adjusted Profilename, generated by autotune. Maybe by taking the inputs profile name with a reference to autotune. Like, in my case. Helena_adjusted. Or Helena_AT.
That would solve the issue.

kristinsmotz, 2025-01-04:
I agree it would be helpful to differentiate in addition to adding kid's name in profile. If space is an issue, maybe it would be possible to replicate the colors of the app icons for AAPSClient vs 2, as background color for the menu or status light sections, or Carb/Calc buttons, or even as the color of the pod/cannula/reservoir icons. Then we'd just need to know which color is associated to which kid, and we would have a hint. There has been several occations I have almost added carbs for the wrong kid. 

Another helpful thing I did was edit the "app name" for the AAPSClient to include kid's name. Per now that requires editing code, so only for advanced users/nerds. I don't know if it would be possible to add a preference option to "include patient name in app name" or "Customized app name".

MilosKozak, 2025-01-04:
https://github.com/nightscout/AndroidAPS/commit/db0974f737516e471f60fdfca87feae0538f1444

MilosKozak, 2025-01-04:
[image]

closed 2025-01-04 by MilosKozak