nightscout/AndroidAPS #3703: Dynamicisf not starting

FalconEagleTest, 2025-01-05:
Hi
I updated from 3.2 to 3.3 two days ago( uninstall 3.2 to then install 3.3 and imported setting), now I always get the message :
Returning to smb not enough data, and returning to profile sensitivity not enough data reason tdd miss, any idea why? 
Thanks
[AndroidAPS_LOG_1736052945398.log.zip](https://github.com/user-attachments/files/18309200/AndroidAPS_LOG_1736052945398.log.zip)

olorinmaia, 2025-01-05:
Hi. To avoid this you should update, not uninstall and install.

Check Statistics page, do you have 7 days of TDD there? If not you need to do a full sync with NSClientV3. Make sure you enable required settings to download bolus-data.

If you aren't able to download data you can disable Dynamic Sensitivity in SMB plugin and activate it again when you got 7 days of TDD data.

FalconEagleTest, 2025-01-05:
i am using NSClientV1 ,
any reason to change to NSClientV3?
thanks.
@olorinmaia

olorinmaia, 2025-01-05:
If you got 7+ days of TDD V3 will be able to get that back with full sync. V1 only takes two days I think.

FalconEagleTest, 2025-01-05:
ok so i changed to v3 and did a full sync, now i see 7days in the Statistics , but still get the same message
[AndroidAPS_LOG_1736101856336.log.zip](https://github.com/user-attachments/files/18311273/AndroidAPS_LOG_1736101856336.log.zip)
@olorinmaia

olorinmaia, 2025-01-06:
What data is missing? Basal/ TBR?

FalconEagleTest, 2025-01-06:
It seems like I am getting the full data since in the statistics I can see data from 30 days ago, but when I look in the log or in the smb window I see tdd24h=null all of those are null

olorinmaia, 2025-01-06:
That sounds strange indeed. Can you provide a screenshot of SMB plugin log and Statistics for TDD + logs? 

I see from your earlier logs that you only missed 7days tdd.
`20:19:46.007 [DefaultDispatcher-worker-6] D/CONSTRAINTS: [ConstraintObject.set():28]: Setting value false -> true (tdd1D=24.349999999999998 tdd7D=null tddLast4H=4.158333333333333 tddLast8to4H=4.766666666666666 tddLast24H=21.929166666666674)[ConstraintObject]`

olorinmaia, 2025-01-06:
If you are comfortable with it, try the following.
1) Check NSClientV3 settings that all sync options are enabled to download bolus, carbs, etc. (just to be sure, you can disable what you don't need after)
2) Maintenance > Export settings (In case of trouble)
3) Maintenance > Reset databases
4) Exit AAPS
5) Start AAPS
6) Do a full sync with NSClientV3

I just did this on my test-rig, and recovered all TDD-data, carbs, tbr etc.

FalconEagleTest, 2025-01-06:
What you saw in the previous log was used for 3 days. So the one thing missing was the 7 days.
Yes if you dotl the full sync it seems to recover everything, but if you turn on the dynamicisf you will see that you will get an error saying tdd7d=null tdd24h=null etc.
@olorinmaia

FalconEagleTest, 2025-01-06:
Yes!!! 
Turning everything on when syncing did the trick , thank you so much for your help @olorinmaia

closed 2025-01-06 by FalconEagleTest