nightscout/AndroidAPS #3704: feature request - Improve priming process of Dana pumps for blind people!

CaroGo, 2025-01-05:
I would like to ask for the option to insert the priming data via AAPS instead of doing it on the pump. Priming and tube filling is a challenge on the Dana pump,if you are blind, as you can't see the display. If the autosetter, which sends filling information to the pump, gets damaged it gets really difficult. It would be comfortable to be able to insert the priming amount manually in aaps (for reservoir and tube), just as it is for priming cannula etc. in actions tab. Thanks in advance!

MilosKozak, 2025-01-05:
Dana pump doesn't support this functionality

closed 2025-01-05 by MilosKozak

CaroGo, 2025-01-05:
Does the pump need to? If you fill the pump and tell the amount aaps this should work as it works now,wouldn't it? You can do cannula and tube filling via aaps...isn't that a similar procedure? I guess it is a pump history issue then...😥

MilosKozak, 2025-01-05:
You can't do it in AAPS, it's faked by doing bolus
I don't know the protocol of autosetter