nightscout/AndroidAPS #3705: Feature request : show loop status on AAPSClient

emilisev, 2025-01-06:
Would be really good to show the loop status on AAPSClient, the same as on AAPS. When AAPS reverts to open loop for some reason, the parent has no way to know. It has happened to me a few times and every time it took me a while to figure out that it reverted to open loop.

reijoahola-nightscout, 2025-01-06:
If this is done, the display will be almost identical with AAPS app. Could there be then something that clearly make these (looking) different and prevent confusion ?

painbrain81, 2025-01-07:
excellent idea. in reality, however, it may seem identical to the master. an icon could be implemented that shows the status of the loop by clicking on the blood sugar or on the clock which in apsclient replaces the "original" loop. this way it would be different at first glance.

emilisev, 2025-01-07:
AapsClient was always different with the additional AAPS/pump status bar. With the latest version, it is even more visible with a specific BG color...

frice79, 2025-01-08:
To day the only way to check the status of the loop on the AAPS Master from remote is:
- send an SMS request to check loop status 
- on the smartwatch, associated to the AAPS Master, that we maintain at home when our son is at school

It happened once the support teacher inadvertently disabled the closed loop of my son. 
We noticed it once our son came home from school with high BG values.
From AAPSClient it would be immediately visible if someone changes the closed loop, it would be a very useful enhancement.
Then in the case of being able to re-enable it, either the possibility is given from AAPS Client or it can remain as an SMS

kristinsmotz, 2025-01-17:
Another way to alert parents to loop status change could be to always add a note in these cases (similar to AAPS Started note), "Closed loop activated", "Open loop activated", "LGS activated" etc. Same with suspend/disconnect. That would make it easier to notice in AAPSClient, and also log the timestamp so we know how long ago it happened. Currently the way I usually notice AAPSClient is from looking at basal rendering in graph, where the line would be completely flat (using only profile basal), and area filled with darker blue instead of the light blue representation of temp basals. There will also be no SMB triangles in the graph for the same period, which hardly ever happens while loop is running for my kids.

olorinmaia, 2025-06-10:
This is now implemented in dev with Running mode that syncs with AAPSClient.

olorinmaia, 2026-09-23:
Now possible in master.

closed 2026-09-23 by olorinmaia