nightscout/AndroidAPS #3707: [3.3.0] Profile display / Basal profile, y-axis wrong values

reijoahola-nightscout, 2025-01-06:
Can be seen both in 3.3.0. master and lates dev build.
When opening Basal display in Profile, y-axis values are clearly shown wrongly.

However, this is only a display problem and does not cause problems when AAPS use Basal.

quizzmaster, 2025-01-06:
For me the y-axis of the basal graph shows plausible values with 3.3.0 master. Do you have a screenshot of the wrong values?

reijoahola-nightscout, 2025-01-06:
There is something strange with these profile pictures and y-axis. The situation seems to change all the time. I have to study this more before posting screenshots.

reijoahola-nightscout, 2025-01-07:
I cannot repeat the problem anymore. Probably disappeared when I played with different fonts and font sizes. May be better to close the case.

closed 2025-01-07 by MilosKozak