nightscout/AndroidAPS #3718: Unpair broken Equil - Issue

CezaryKlose, 2025-01-08:
My insulin pump Equil was paired with the AAPS 3.3.1.0 Build: c7e8f62f75-2025.01.08
Pump stopped responding at all and Equil provided replacement.
But I'm unable to unpair pump since there is no connection possibility to establish.
[AndroidAPS._2025-01-08_05-24-11_.26.zip](https://github.com/user-attachments/files/18344805/AndroidAPS._2025-01-08_05-24-11_.26.zip)

Any help or advise would be appreciated.

MilosKozak, 2025-01-08:
only master version is supported here

closed 2025-01-08 by MilosKozak

CezaryKlose, 2025-01-08:
Rebuilded application with full github synchronization.
Version is AAPS 3.3.1.0 Build: 2ac634c908-2025.01.08
Tried again to unpair pump but also connection error and it is still paired.

I have attached logs and screenshots.

Please kindly verify what may be wrong.

Thank you.
[image]
[AndroidAPS._2025-01-08_11-54-35_.42.zip](https://github.com/user-attachments/files/18346384/AndroidAPS._2025-01-08_11-54-35_.42.zip)
[image]