nightscout/AndroidAPS #3719: [3.3.1.0] DynISF calculates values outside of limit

KeesV, 2025-01-08:
I'm having an issue with DynISF after upgrading to 3.3.1.0 from 3.2.0.4.
When I turn on DynISF and the loop runs, I get an error message saying "Profile sensitivity value is outside of hard limit. INSANE VALUE limited to 1000,00".
I looked at the logs and found a huge amount of lines like this:
`09:00:21.022 [DefaultDispatcher-worker-2] D/APS: [OpenAPSSMBPlugin.getAverageIsfMgdl():167]: getAverageIsfMgdl() null from 0 values 07-01-2025 03:45:00 iobCobOref1Worker.`

When I turn off DynISF everything runs fine.

[AndroidAPS.log](https://github.com/user-attachments/files/18345544/AndroidAPS.log)

Build version: `2ac634c908-2025.01.07`
Time of occurrence in the logfile: `09:35:39.486`

Note that I tried turning DynISF on and off a couple of times this morning. First time of occurrence was `09:00:18.622`.

What I have tried (but unfortunately didn't fix it):
- Leave AAPS running overnight (with DynISF) disabled and try turning DynISF on again in the morning.
- Erase APS results from maintenance menu and restart AAPS

[image]

MilosKozak, 2025-01-08:
Send me printscreen of TDD from statistics page and then try to reset it there and restart AAPS

KeesV, 2025-01-08:
OK, steps I took:
- Take screenshot of TDD statistics (see below)
- Recalculate TDD statistics by pressing the button "Herberekenen"
- Als pressed the "Reset" button at the bottom of the page, just to be sure
- TDD statistics were recalculated by AAPS. Numbers were exactly the same as in the screenshot
- Restarted AAPS
- Enabled DynISF (I had it off before because of this issue)
- Result: Same error is shown...

[image]

MilosKozak, 2025-01-08:
are you able to build dev branches?

KeesV, 2025-01-08:
I should be able to. I don't have a separate phone to test however, so as long as I can safely switch back to main it should be good

MilosKozak, 2025-01-08:
i'll prepare more logging version

MilosKozak, 2025-01-09:
use dev branch a put new log here once this issue appears again

KeesV, 2025-01-10:
Thanks @MilosKozak ! The issue is solved 👍 

The problem was that my DynISF adjustment factor was set to 0. I found this by digging through the log from the dev branch and through the code. I saw that `tdd=0.0` and when looking at how that was calculated I saw that it includes 
```* preferences.get(IntKey.ApsDynIsfAdjustmentFactor)```
so I checked that and it was set to 0.

Setting my DynISF adjustment factor to 100 made the issue go away and all is running smoothly again.

However, I don't understand how it got to 0. The UI doesn't let me enter 0 as a value. That might be worth looking into.

Attaching the log that led me to the solution here.
[AndroidAPS.log](https://github.com/user-attachments/files/18372109/AndroidAPS.log)