nightscout/AndroidAPS #3722: [3.3.1.0] AAPS app Status light / Threshold warning pump battery level (yellow) does not work

reijoahola-nightscout, 2025-01-09:
AAPS app Status light / Threshold warning pump battery level (yellow) does not work.
It stays white all the time.

Noticed at least in AAPS 3.3.1.0 build but may it is already in 3.3.0.0

reijoahola-nightscout, 2025-01-09:
Screenshots are from test system (virtual pump) where pump battery level is continuously 50%.
Originally the problem was noticed wit real pump when A-C Insight battery level lowered into 95% where I has the yellow warning.
[image]
[image]
[image]

MilosKozak, 2025-01-09:
fixed by https://github.com/nightscout/AndroidAPS/commit/0c3e1ad156e3ed7501be92596f7b324b72e4a3c8

closed 2025-01-09 by MilosKozak