nightscout/AndroidAPS #3723: Persistent problem. V3+ with enabled websockets / no synchronization with client.

elgrupo123, 2025-01-09:
I have already mentioned the problem several times. 3.3.1.0 does not fix the error either. As soon as you use several AAPS clients and all have activated websockets, only 1 device runs properly for a maximum of 12 hours, after which it behaves exactly like the other follower devices, which cannot synchronize the BG values ​​with active websockets from the start. All other data arrives. Even a new, clean installation did not help. The solution is - master (websockets activated), follower (websockets deactivated).

elgrupo123, 2025-01-09:
[AndroidAPS.log](https://github.com/user-attachments/files/18361436/AndroidAPS.log)

olorinmaia, 2025-01-09:
Strange. I remember this in dev. We are using 3 phones with AAPSClient. All websockets, no issues so far. 10be.de master NS branch

elgrupo123, 2025-01-10:
> What happens if you pause / unpause in NSClientv3 plugin?

Nothing. I've already tried everything. And as I said, there are no problems with 3.2.0.4.

kristinsmotz, 2025-01-10:
@MilosKozak I sent you a PM with nightscout link and token for one of my sites. Here are also a couple screenshots from the other kid illustrating that AAPSClient gets some info, enough to display SMB's and predictions in graph, but no BG values:
[image]
[image]

logs from AAPSClient: 
[AndroidAPS_LOG_1736537594947.log.zip](https://github.com/user-attachments/files/18381511/AndroidAPS_LOG_1736537594947.log.zip)
I'll pm you the site + token for this site as well

kristinsmotz, 2025-01-10:
Since I am very nerdy today, i dug into the log and found some clues. 

The last glucose value was visible in AAPSClient at 19:58:32, first reference in log is here, where it collects entries, does a bunch of tricks, I highlighted inserting BG value.

19:58:32.921 [EventThread] D/NSCLIENT: [NSClientV3Service.onDataCreateUpdate$lambda$14():216]: onDataCreateUpdate: {"colName":"entries","doc":{"app":"AAPS","date":1736535512757,"device":"Libre3","direction":"Flat","filtered":0,"isReadOnly":false,"isValid":true,"sgv":163,"type":"sgv","unfiltered":0,"units":"mg\/dl","utcOffset":60,"created_at":"2025-01-10T18:58:32.757Z","identifier":"a3818769-f5e8-5b2d-a1cc-d48532ae35a1","srvModified":1736535513257,"srvCreated":1736535513257,"subject":"Kristin"}}
19:58:32.924 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():262]: ◄ WS CREATE/UPDATE entries <i>{"app":"AAPS","date":1736535512757,"device":"Libre3","direction":"Flat","filtered":0,"isReadOnly":false,"isValid":true,"sgv":163,"type":"sgv","unfiltered":0,"units":"mg\/dl","utcOffset":60,"created_at":"2025-01-10T18:58:32.757Z","identifier":"a3818769-f5e8-5b2d-a1cc-d48532ae35a1","srvModified":1736535513257,"srvCreated":1736535513257,"subject":"Kristin"}</i>
19:58:32.927 [EventThread] D/NSCLIENT: [NsIncomingDataProcessor.processSgvs():102]: Received NS Data: [NSSgvV3(date=1736535512757, device=Libre3, identifier=a3818769-f5e8-5b2d-a1cc-d48532ae35a1, srvModified=1736535513257, srvCreated=1736535513257, utcOffset=60, subject=Kristin, isReadOnly=false, isValid=true, sgv=163.0, units=MG_DL, direction=FLAT, noise=null, filtered=0.0, unfiltered=0.0)]
19:58:32.946 [RxCachedThreadScheduler-63] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():62]: Firing EventNewBG GlucoseValue(id=284815, version=0, dateCreated=1736535512930, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=a3818769-f5e8-5b2d-a1cc-d48532ae35a1, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1736535512757, utcOffset=3600000, raw=0.0, value=163.0, trendArrow=FLAT, noise=null, sourceSensor=LIBRE_3)
19:58:32.947 [RxCachedThreadScheduler-63] D/DATABASE: [CompatDBHelper$dbChangeDisposable$2.accept():67]: Firing EventNewHistoryData 1736535512757 GlucoseValue(id=284815, version=0, dateCreated=1736535512930, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=a3818769-f5e8-5b2d-a1cc-d48532ae35a1, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1736535512757, utcOffset=3600000, raw=0.0, value=163.0, trendArrow=FLAT, noise=null, sourceSensor=LIBRE_3)
19:58:32.948 [RxCachedThreadScheduler-63] D/DATABASE: [PersistenceLayerImpl$insertCgmSourceData$4.apply():586]: Inserted GlucoseValue from NSClient GlucoseValue(id=284815, version=0, dateCreated=1736535512930, isValid=true, referenceId=null, interfaceIDs_backing=InterfaceIDs(nightscoutSystemId=null, nightscoutId=a3818769-f5e8-5b2d-a1cc-d48532ae35a1, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1736535512757, utcOffset=3600000, raw=0.0, value=163.0, trendArrow=FLAT, noise=null, sourceSensor=LIBRE_3)
**19:58:32.949 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():262]: ◄ INSERT GlucoseValue 1**
19:58:32.951 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():262]: ● RUN Starting upload NEW_DATA
19:58:32.989 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():262]: ► UPL Start
19:58:33.005 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():262]: ► UPDATE food Sent PairFood 66816928f5a8750002de114e <i>{"carbs":0,"category":"","date":1736535513004,"energy":0,"fat":0,"gi":2,"identifier":"66816928f5a8750002de114e","isReadOnly":false,"isValid":true,"name":"","portion":0.0,"protein":0,"subCategory":"","unit":"g"}</i> 59/86
19:58:33.318 [DefaultDispatcher-worker-8] E/NSCLIENT: [NSClientV3Plugin.dbOperationFood():590]: Upload exception
java.lang.IllegalArgumentException: URL "v3/food" does not contain "{identifier}". (parameter #2)
    for method NightscoutRemoteService.updateFood
	at retrofit2.Utils.methodError(Utils.java:56)
	at retrofit2.Utils.methodError(Utils.java:44)
	at retrofit2.Utils.parameterError(Utils.java:68)
	at retrofit2.RequestFactory$Builder.validatePathName(RequestFactory.java:843)
	at retrofit2.RequestFactory$Builder.parseParameterAnnotation(RequestFactory.java:427)
	at retrofit2.RequestFactory$Builder.parseParameter(RequestFactory.java:335)
	at retrofit2.RequestFactory$Builder.build(RequestFactory.java:213)
	at retrofit2.RequestFactory.parseAnnotations(RequestFactory.java:67)
	at retrofit2.ServiceMethod.parseAnnotations(ServiceMethod.java:26)
	at retrofit2.Retrofit.loadServiceMethod(Retrofit.java:235)
	at retrofit2.Retrofit$1.invoke(Retrofit.java:177)
	at java.lang.reflect.Proxy.invoke(Proxy.java:1006)
	at $Proxy5.updateFood(Unknown Source)
	at app.aaps.core.nssdk.NSAndroidClientImpl$updateFood$2.invokeSuspend(NSAndroidClientImpl.kt:416)
	at app.aaps.core.nssdk.NSAndroidClientImpl$updateFood$2.invoke(Unknown Source:8)
	at app.aaps.core.nssdk.NSAndroidClientImpl$updateFood$2.invoke(Unknown Source:2)
	at app.aaps.core.nssdk.NSAndroidClientImpl$callWrapper$2$1.invokeSuspend(NSAndroidClientImpl.kt:496)
[... cut ...]
20:03:32.026 [EventThread] D/NSCLIENT: [NsIncomingDataProcessor.processSgvs():123]: Ignoring record with wrong timestamp: GV(id=0, version=0, dateCreated=-1, isValid=true, referenceId=null, ids=IDs(nightscoutSystemId=null, nightscoutId=3274af65-119b-538f-974d-a5671e4adf2f, pumpType=null, pumpSerial=null, temporaryId=null, pumpId=null, startId=null, endId=null), timestamp=1736535812282, utcOffset=3600000, raw=0.0, value=138.0, trendArrow=FORTY_FIVE_DOWN, noise=null, sourceSensor=LIBRE_3)
20:03:33.029 [RxCachedThreadScheduler-47] D/NSCLIENT: [NSClientV3Plugin$onStart$10.accept():262]: ● DONE PROCESSING BG

MilosKozak, 2025-01-10:
I see Pump status update but not BG. Can you confirm you upload BGs using v3 protocol?

kristinsmotz, 2025-01-10:
I've enabled upload BG to NS in kid's phones AAPS. All phones using NSClientv3, both in AAPS and AAPSClient/2

elgrupo123, 2025-01-10:
It's very similar for us. Two M5 stacks and several Garmin watches get the BG values ​​from NS, and there are no dropouts. I now switch the websockets OFF on the followers (it is active on the master), otherwise the NS client cannot be used properly. You can also send the NS page + password. PM to Facebook??

MilosKozak, 2025-01-10:
@kristinsmotz you don't have synchronized time on all phones. Set Android for automatic time synchronization

elgrupo123, 2025-01-11:
I activated websockets on my phone overnight and as expected there was no BG sync after a few hours. The last BG value came through at 2:08AM. Devices with deactivated websockets ran without problems.
[AndroidAPS._2025-01-11_00-00-16_.0.zip](https://github.com/user-attachments/files/18384927/AndroidAPS._2025-01-11_00-00-16_.0.zip)
[image]
[image]

MilosKozak, 2025-01-11:
expecting you'll have the same issue

elgrupo123, 2025-01-11:
> expecting you'll have the same issue

All phones have automatic time synchronization. That really can't be the problem. And as I said, if I downgrade to 3.2.0.4, it works perfectly with activated websockets. Just tried it again.

MilosKozak, 2025-01-11:
pm me ns address and token

elgrupo123, 2025-01-11:
> pm me ns address and token

is out via Facebook

MilosKozak, 2025-01-11:
i see no message pending on FB, what's your name there

MilosKozak, 2025-01-11:
it's the same issue. BGs are comming 0,5sec in the future, which doesn't pass validation

elgrupo123, 2025-01-11:
> i see no message pending on FB, what's your name there

Katinka Mina

elgrupo123, 2025-01-11:
> it's the same issue. BGs are comming 0,5sec in the future, which doesn't pass validation

and how can the problem be solved?

MilosKozak, 2025-01-11:
Did you check time synchronization again?
how do you get BGs to AAPS?

elgrupo123, 2025-01-11:
> Did you check time synchronization again? how do you get BGs to AAPS?

Time Synchronisation Automatic.
Juggluco-xdrip-aaps-NS.

ricoin1, 2025-01-11:
Facing the same isssue, but using Byoda-AAPS-NS.

MilosKozak, 2025-01-11:
then it's necesssary to check in juggluco and xdrip where is the source of timestamp in the future

kristinsmotz, 2025-01-11:
All phones have used automatically time and time zone here. I've tried turning this one off now on my phone to test.
[image]

elgrupo123, 2025-01-12:
Thank you @MilosKozak  Everything is working perfectly with the development!