nightscout/AndroidAPS #3724: Version Checker not working on clean install of 3.2.0.4

jbfeutz, 2025-01-09:
I had to install AAPS on a backup phone due to damage to my primary phone. I installed the same APK I had been using (my own fork of autoISF on 3.2.0.4). After setup I was unable to get off of LGS and I noticed in the constraints that Version Checker was indicating an old version. I had limited time to get it to work so I removed the version checker code from my fork and was able to get it to close the loop. 

To rule out any of my modifications I built a clean master version of 3.2.0.4 and installed on a different phone and am having the same issues. Screenshots and logs attached.

I found one other comment on discord of someone having the same issue back in June: https://discord.com/channels/629952586895851530/629954570394533889/1251826771532320778
[AndroidAPS_LOG_1736444630682.log.zip](https://github.com/user-attachments/files/18365242/AndroidAPS_LOG_1736444630682.log.zip)
[image]
[image]
[image]

MilosKozak, 2025-01-09:
3.2 is no more supported
upgrade to latest version

closed 2025-01-09 by MilosKozak

jbfeutz, 2025-01-09:
I understand that upgrading is a solution to this issue, but I believe the root cause should still be fixed. Ideally others can confirm this, but my experience has been that you cannot install on old version (even if it is still in the "usable window") once a new version is available. This would mean that you cannot quickly restore from a backup if a new version has been released since your last backup. I believe this to be a major safety concern and something that should be looked in to further.

jbfeutz, 2025-01-10:
Update in case others find this... the issue seems to be isolated to a single preferences file I was using as a backup. When I use a newer preferences file it allows me to close the loop but the older file, when imported, breaks it again. A discord user also tested this with an old preferences file and they DID NOT have the same issue so it appears to be isolated to this specific preferences file somehow.