nightscout/AndroidAPS #3732: Unable to change DynamicISF Adjustment Factor % Value

spencertollefson, 2025-01-11:
Build: 3.3.1.0

I appear unable to adjust the `DynamicISF Adjustment Factor % Value` value found in `OpenAPS SMB` preferences. After I input a new value and click "OK", it continues to show the old value. Please see video for a clear understanding of this issue.

I believe this issue is new after I upgraded from 3.2.0.4 to 3.3.1.0

https://github.com/user-attachments/assets/fe671f75-7fad-405d-b151-fd1f15b17180

markvm15, 2025-01-12:
Same problem

olorinmaia, 2025-01-12:
I'll share a workaround i know have worked for others having this issue until other solutions are found. I can't guarantee that it will work so do it only if you feel it's worth the risk. 

1. You need to have backup of data at Nightscout and can use NSClientv3 to restore data with full sync after. 
2. !! Backup settings in Maintenance > export settings !!
3. Uninstall AAPS
4. Install AAPS
5. Import settings
6. Do a full sync with NSClientv3

markvm15, 2025-01-12:
> I'll share a workaround i know have worked for others having this issue until other solutions are found. I can't guarantee that it will work so do it only if you feel it's worth the risk.
>
> 1. You [...]

This work for me, Really Thx.

spencertollefson, 2025-01-12:
Also worked for me. Thank you @olorinmaia!

PoweRGbg, 2025-01-13:
Same problem here. After going back to 3.3.0.4 I was able to change it.

spencertollefson, 2025-01-13:
Sorry @MilosKozak  - I didn't look at nor save any logs while I had the issue.

MilosKozak, 2025-01-13:
anyone else?

MilosKozak, 2025-01-14:
can you re-test with https://github.com/nightscout/AndroidAPS/commit/7c000911068903ed7477c1fcb9d80dc88a893baf ?

closed 2025-01-14 by MilosKozak