nightscout/AndroidAPS #3737: [Dana-i] Large bolus causes endless 'Waiting for estimated bolus end' without actually applying bolus

wouterlemcke, 2025-01-13:
When I do reasonably large boluses (>15 U) AAPS regularly fails to apply the bolus. 

I am now running version 3.3.1 on Google Pixel 8 (Android 15), but it was in earlier versions and older phones as well.

When I try to apply the bolus it directly goes to 'Waiting for estimated bolus end' and will stay there for a long time.
I can also not stop the bolus anymore. Clicking the bolus will make the popup go to 'STOP PRESSED' and from there I cannot do anything anymore. I have to kill the app to continue, manually applying the bolus on my pump.

See the video

https://github.com/user-attachments/assets/a64fe82c-722e-4e88-8805-777175dc1fa5

I am happy to share more info, but not sure what you need

MilosKozak, 2025-01-13:
do such bolus and provide logs and time

wouterlemcke, 2025-01-18:
Where can I find the logs?

MowgliChild, 2025-01-19:
> Where can I find the logs?

Send log file vvia email when this happens.
Hamburger, maintenance, email log file. Record time and date and add that here so the Devs can find it.

wouterlemcke, 2025-01-20:
Thanks! I send the logs. 

10:08:24 was time when I tried the bolus.

Error I see is: `10:08:28.691 [binder:29854_F] E/CORE: [DanaRSPacketBolusSetStepBolusStart.handleMessage():43]: Result Error: 128`

wouterlemcke, 2025-01-20:
Ah, check. I will investigate there. Thanks for the quick reply!

wouterlemcke, 2025-01-20:
Hmm..

Max bolus is 50, max daily is 130 (right now at 21..).
I think it's the safety ratio, that's set to 100. I will turn it off.

For now I think this issue can be closed. I will reopen if this is not the solution 😉

Thanks for the help!

closed 2025-01-20 by wouterlemcke

MilosKozak, 2025-01-20:
If this happen again try to issue the same bolus directly on pump