nightscout/AndroidAPS #3739: Medtrum Nano Preferences: Hourly and Daily Maximum Insulin settings do not persisted on modfication

vanelsberg, 2025-01-13:
**AndroidAPS 3.3.1.0-dev**
Build: f751b7658c
Flavor: fullrelease
Pump: Medtrum Nano

**Problem**
Pump driver Medtrum Preferences do not allow modifying the **Hourly and Daily Maximum Insulin settings**
User does get to option to change, modified value is not persisted however.

[image]

vanelsberg, 2025-01-14:
Logfile, UI test sequence:
1) 09:46:42.051: Export preferences (maintenance)
2) MT tab -> Metrum Preferences
3) Change Hourly -> 12, Change Daily Max -> 45
4) MT tab: refresh

Can not find anything in the log that could relate to the problem?

[AndroidAPS.log](https://github.com/user-attachments/files/18407960/AndroidAPS.log)

MilosKozak, 2025-01-14:
resolved by https://github.com/nightscout/AndroidAPS/commit/7c000911068903ed7477c1fcb9d80dc88a893baf

closed 2025-01-14 by MilosKozak

vanelsberg, 2025-01-14:
Confirming 👍 
With this preference changes  are persisted.