nightscout/AndroidAPS #3746: [3.3.1.2] Issue with Bluetooth Devices List  "TWI Board"

m4nnihuber, 2025-01-15:
after Updating i did a pod change. now there are 2 "TWI Board" in my Bluetooth device list. 
Looks like the Old Pod and The New Pod are listed but Name is identically.

How is the right workflow on a podchange? 
- deactivate pod
- delete TWI Board from Bluetooth Devices
- start new pod?

otherwise i have a lot of twi Board devices in my list. 
Maybe tgqt causew Problems like when having a lot of Dexcom Devices in the List?

Thanks for your Support

vanelsberg, 2025-01-15:
AAPS will unbind the pod on de-activation.
I suspect the old "TWI Board" is a manually bounded Pod from before upgrading AAPS?
You should be able to remove using unpair/forget device.

m4nnihuber, 2025-01-16:
I never got a promt for connection with "TWI Board" before Update to 3.3.1.2 
after update the promt came up and i connected.
Same happened after changing pod.

Since the second pod i also have 2 identically bluetooth devices in the list.
buth both have different bluetooth adresses

vanelsberg, 2025-01-16:
> Since the second pod i also have 2 identically bluetooth devices in the list. buth both have different bluetooth adresses
I think that is because DASH does not present a unique name for the device. That was not needed as it was not designed to be visible in the list of BT devices.

When bonding your phone "remembers" the specific DASH device dat was connected. Similar to the Dexcom transmitters after change. Phone will probably cleanup after some time. I'd say you can safely ignore.

olorinmaia, 2025-01-16:
Did you have connection issues with Dash on android 15 before upgrading to 3.3.1.2?

m4nnihuber, 2025-01-17:
Not before aupdate. But now there are serveral Issues 
Mostly Notification TBR could not be set or unknown Error 01

vanelsberg, 2025-01-18:
See also [Can't connect to Omnipod DASH pods after Android 15 upgrade](https://github.com/nightscout/AndroidAPS/issues/3471#issuecomment-2599803223)

MilosKozak, 2025-01-19:
https://github.com/nightscout/AndroidAPS/commit/8abfbd51842f6f9344f5764bd93f5a78113753a0

closed 2025-01-19 by MilosKozak