nightscout/AndroidAPS #3749: 3.3.1.2 Continual Bluetooth Pairing Requests

32423676573, 2025-01-16:
Using Omnipod Dash. The Bluetooth Pairing Requests started mid-pod as soon as I upgraded from. 3.3.1.0 to 3.3.1.2.

Each time AAPS connects to the pod I receive a series of 3 pairing requests, with the device matching the MAC of the pod.

Accepting the pairing request breaks connectivity between AAPS and the pod. Once the pod is unpaired the communication works.

In this state every few minutes my phone is flooded with pairing requests as AAPS sends instructions to the pod, or each time I change a setting.

Will upload logs.

olorinmaia, 2025-01-16:
I'm guessing your on android 15. Did you have issues with Dash before upgrading to 3.3.1.2?

32423676573, 2025-01-16:
Hey Alf,
You're right, My device is a OnePlus 11 5G on oxygen OS 15 (Android 15).
I've confirmed it's 3.3.3.2 specific as I've rolled back to 3.3.1.0 and the Bluetooth prompts went away.
Let me know if there's anything else I can provide to assist.

"target": "https://github.com/nightscout/AndroidAPS/issues/3749#issuecomment-2597134797",
"url": "https://github.com/nightscout/AndroidAPS/issues/3749#issuecomment-2597134797",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

olorinmaia, 2025-01-16:
Hi again! I would like to know if you got a stable connection to dash with android 15 in 3.3.1.0😉

32423676573, 2025-01-16:
You got it. No problems at all with Android 15 on 3.3.1.0

olorinmaia, 2025-01-17:
@MilosKozak see my comment in https://github.com/nightscout/AndroidAPS/issues/3471#issuecomment-2592338943

I think the best way forward is to revert this change for Dash in Master branch or keep it in a separate branch. If you want to keep it in Master we should make it an option in Dash settings defaulted off. 

I got a strong feeling that for many it will make things worse. And for the ones having issues they can try enabling option for manual bonding.

m4nnihuber, 2025-01-17:
I am having same issues regarding Android 15.
see Issue #3746

mjhuisman, 2025-01-17:
Here same problem: Android 15 with dash at 3.3.1.0 no issues, but at 3.3.1.2 i have to bond pod, but after bonding its impossible for aaps to connect.
Im using Xiaomi Redmi note 13 pro.

m4nnihuber, 2025-01-18:
Looks like it gets way better when i delete old "twi boards" in bluetooth list before starting new pod

Not-AriStienfeld, 2025-01-18:
Same issue, it goes away after unpairing and repairing but comes back after maybe 10 minutes

MilosKozak, 2025-01-19:
https://github.com/nightscout/AndroidAPS/commit/8abfbd51842f6f9344f5764bd93f5a78113753a0

closed 2025-01-19 by MilosKozak