nightscout/AndroidAPS #3754: (dev 3.1.1.2) Bug in Preferences from AutoISF-Plugin

Josch24, 2025-01-17:
The value of “variable SMB delivery ratio. mapped glucose range” changes automatically to a very high value, if you set value <39. To reproduce the error, please enter a value <39 and exit the settings. When you call up the settings again, the high value will be displayed.  

[image]
[image]
[image]

ga-zelle, 2025-01-19:
I got the impressin that there is a recent problem with UnitDouble Keys. Until clarified/solved enter those "smaller" ranges as mmo/L numbers without actually selecting mmol/L as your units.

Gohtraw, 2025-01-26:
38 * 18 = 684
It  Looks like it converts the entered range from mmol/l to mg/dl

ga-zelle, 2025-02-22:
Ready to be closed; see above

closed 2025-02-22 by MilosKozak