nightscout/AndroidAPS #3755: invalid pump history detected

Volberger, 2025-01-17:
[AndroidAPS.log](https://github.com/user-attachments/files/18456975/AndroidAPS.log)

[AndroidAPS._2025-01-17_01-00-00_.8.zip](https://github.com/user-attachments/files/18456998/AndroidAPS._2025-01-17_01-00-00_.8.zip)

olorinmaia, 2025-01-17:
Can you add screenshots of what the problem is and abit more info?

Volberger, 2025-01-19:
I have received this error message frequently in the last few years. I did not notice that the pump function was somehow restricted. The "sleep" option is offered. When the button is pressed, the error message disappears and the loop continues to work normally. Sometimes after weeks, sometimes after days, the error appears again. I use an older Medtronic pump model 712 in combination with RileyLink.

[image]

[image]

[AndroidAPS._2025-01-19_01-00-00_.5.zip](https://github.com/user-attachments/files/18468470/AndroidAPS._2025-01-19_01-00-00_.5.zip)
[AndroidAPS._2025-01-19_01-00-00_.6.zip](https://github.com/user-attachments/files/18468471/AndroidAPS._2025-01-19_01-00-00_.6.zip)
[AndroidAPS.log](https://github.com/user-attachments/files/18468472/AndroidAPS.log)

Volberger, 2025-02-10:
> Can you add screenshots of what the problem is and abit more info?

I really don't know what further information you want. I have already provided screenshots, logs and error descriptions. Again: the error message appears irregularly but always. The program stability and scope are not restricted, the loop runs well. The error is simply that an error is claimed (invalid pump history) but there is obviously no error, at least no error that I as a user could notice.

andyrozman, 2025-02-10:
This error comes when certain values can't be determined (TBR duration). 

In past main culprit of this was wrong decoding of history, but it seems that in this case history is ok, and problem is somewhere else... 

We would have to look what is happening there that this happens... Problem is that I don't have time to do it...

MilosKozak, 2026-10-16:
Thank you for the logs. They show the cause, and it is fixed in b2d21ce679 on `dev`.

**What happened (log of 2025-01-19):**

- The history read at 10:49:56 brought the entries from 10:44:54 on, but not three older ones: a cancel at 10:43:53, a temp basal at 10:43:55 and a cancel at 10:44:52.
- The read at 10:56:41 brought these three as new entries. The oldest of them is a cancel.
- For a cancel, AAPS takes the temp basal before it from the earlier reads, because that is the one the cancel ends. But it took the *newest* temp basal from the earlier reads (10:47:56), not the one before the cancel (10:41:55).
- So the temp basal ran from 10:47:56 to 10:43:53, which is -243 s. AAPS skipped it and showed "Invalid pump history data detected".

This is why the message came back every few days while the loop kept working.

**The fix:** AAPS now takes the newest earlier entry that is older than the late entry. In this log that is the temp basal of 10:41:55, and every temp basal gets the right length. The same rule now applies to suspend and resume. There, the same mistake gave no message, but it could record the pump as stopped while it was running.

**Not fixed yet:** why a history read sometimes misses entries and brings them later. With this change AAPS handles them correctly when that happens.

It is tested with the times from your log. We cannot make a real pump miss entries on purpose, so there is no test on a pump. If you still see "Invalid pump history data detected" with a build that has this change, please write here again and attach the logs.

closed 2026-10-16 by MilosKozak

MilosKozak, 2026-10-16:
An update to my comment above: we now also know why the history read missed the three entries, and that part is fixed too.

**In your log, the read at 10:49:56 did this:**

- The pump had just started a new history page, so page 0 had only 16 entries, from 10:44:54 on.
- Page 1 came with a wrong checksum ("getPumpHistory: checksum is wrong"), and the read stopped there.
- AAPS still processed page 0 and remembered 10:47:56 as the last entry. The three newest entries of page 1 (10:43:53 to 10:44:52) were not read.
- The next read looks back 35 minutes from the last entry, so it found them, but too late.

The same log has two more checksum errors (08:28:40 and 09:30:08). They lost nothing, because there page 0 itself failed, or page 1 had no entries that were not read before. Entries are lost only when page 1 fails soon after the pump has started a new page. This is why the message came only every few days.

**Fixed in 50c8e9b40b on `dev` (#3378):** when a page fails, AAPS now processes nothing from that read and keeps the last entry as it was. The next read then gets all entries again.

So there are two fixes now:

- 50c8e9b40b: a bad page does not make entries come late.
- b2d21ce679: if entries still come late for another reason, for example a change of the pump clock, they are paired correctly.

Both are only on `dev` now, not in a release yet.