nightscout/AndroidAPS #3759: [Bug] Omnipod not pairing properly after update

Not-AriStienfeld, 2025-01-18:
Updated yesterday, bug started today when I changed pumps. Updated from 3.2.0.1 to 3.3.1.2, and it prompted me to pair with "TWI Board" which it never has before. Now after a bunch of time, every time it tries to handshake it stops abruptly and disconnects, trying again repeatedly. The only way to temporarily fix it is to unpair and repair TWI Board. 

 https://github.com/user-attachments/assets/4ef0237e-6836-42d4-9313-76cfebc95bef

I'm trying to upload logs but it isn't working

Not-AriStienfeld, 2025-01-18:
Issue is with new Bluetooth bonding, I suggest making an option to use legacy or new bonding under omnipod settings

vanelsberg, 2025-01-18:
Maybe the problem is the unpair/repair process for the TWI device ...

jaredwhittle, 2025-01-18:
Is there an article or comment somewhere that talks about the new bluetooth bonding and why AAPS changed to it? Thanks!

vanelsberg, 2025-01-18:
> Is there an article or comment somewhere that talks about the new bluetooth bonding and why AAPS changed to it? Thanks!

See https://github.com/nightscout/AndroidAPS/issues/3471#issuecomment-2599803223

Not-AriStienfeld, 2025-01-19:
> > Is there an article or comment somewhere that talks about the new bluetooth bonding and why AAPS changed to it? Thanks!
> 
> See [Can't connect to Omnipod DASH pods after Android 15 upgrade](https://github.com/nightscout/AndroidAPS/issues/3471#issuecomment-2599803223)
> 

I tried all this and it still wouldn't work, repeating the same error until eventually twi board stopped showing up in Bluetooth overall. I have a pixel 8 pro running android 15. I tried connecting in an older version but it seems like something it did made it so the pump straight up wouldn't connect anymore. I really think this new bluetooth mode should be toggleable, since I had zero issues with the normal Bluetooth mode.

campbellgl, 2025-01-19:
Pixel 8a, Android 15. Mine worked fine with the existing pod after updating to 3.3.1.2 (asked me to bond first thing and it worked normally). Then I changed the pod, which activated fine but then never connected again (got the loud warnings, etc.). I downgraded back to 3.3.1.0 and it was fine again. (At some point I restarted phone, but can't remember if that was before or after the downgrade.) I tried updating back to 3.3.1.2 and it seems fine again.

MilosKozak, 2025-01-19:
https://github.com/nightscout/AndroidAPS/commit/8abfbd51842f6f9344f5764bd93f5a78113753a0

closed 2025-01-19 by MilosKozak

vanelsberg, 2025-01-19:
Can't test this as I am not on Android 15, and currently not on DASH.
As **bonding is now optional**, can any of you please test and report back here?