nightscout/AndroidAPS #3760: Watch face does not sync from Android Aaps to Watch?

in2961, 2025-01-18:
After installing latest AAPS wear 3.3.1.2 on my Samsung G6 Watch (OneUI 6.0, Wear OS 5.0) via Wear2 app...(.I did delete the previous AAPS app on the watch)....

I somehow cannot re-install/send the custom watch face (AAPS Wear tap, load watch face) via my Samsung S22 Ultra (OneUI 6.0, Android 14)?

When I select the desired watchface... it waits a little time then displays the choosen watchface... but on my watch nothing happens?

Rarely the watchfaces I used before having to switch to the Wear installer are still on my watch and working dromen old AAPS version I guess?

Anyone any advice on how to solve this? I want my chosen watch face to appear on my watch?
Might it be related to another issue on github mentioning MTR93600 Watchface not available

vanelsberg, 2025-01-18:
> When I select the desired watchface... it waits a little time then displays the choosen watchface... but on my watch nothing happens?

On the watch, did you also select the AAPS Custom watchface?

in2961, 2025-01-18:
Thanks for your quick reply! 

I do not think I have/can...not with the watch faces on the AAPS app on my phone and tranfer them from there. See picture one.. something changed here since  uofrade to UI 6.0 on the [watch?](

[image]

.. I did select an onder watchface on my Watch , the one I would not like to use , see.pictire 2 from the watch. 

[image]

vanelsberg, 2025-01-19:
Double check you have installed the AAPS wear .apk that was build with the same AAPS build and same signing key.

You should be able to activate/select select the AAPS(custom) watch face. Then go to wear on the phone to select the desired watchface.

[image]

in2961, 2025-01-19:
Tha k you that is waht solved it!

closed 2025-01-19 by in2961