nightscout/AndroidAPS #3761: Feature request: Add real calculation ability to calculator & carbs input windows

seiphr, 2025-01-19:
Oftentimes, I eat a pre-packaged item that has it's carbohydrates listed per 100g. It would be beneficial if I could enter the per-100g carbs and the actual weight of the item in grams and have AAPS do the calculation for me. Otherwise I have to get a calculator app open or work it out mentally which usually involves rounding.

The same feature should also exist in the carbs window.

olorinmaia, 2025-01-19:
Cool idea. I could imagine a button somewhere on the row where you enter amount of carbs, when touched, a dialogue box that allow you to calculate KH in grams given that you know per 100g and actually weight in g.

painbrain81, 2025-01-19:
Very cool idea! I suggest 100gr default for weight form (you can change of course).

pnharris02, 2025-01-21:
Yes, useful idea.  The 100g default could be changed to the 'typical serving size' if that was what was on the packaging, though usually 100g would be available.
Even better would be if when this dialog box opened, it had multiple lines, and you could add multiple elements of the meal, and AAPS would provide the total carbs at the bottom.  My brain can usually cope with mental maths for one ingredient, but multiples gets a bit more difficult!

pnharris02, 2025-01-22:
I was nearly going to suggest that this data could be pulled from a foods database too, but didn't want to make my request too overwhelming for developers in the first instance.
I've just found another older open issue requesting a similar feature though, initiated from the foods database
https://github.com/nightscout/AndroidAPS/issues/1105