nightscout/AndroidAPS #3762: [Bug] Carbs suggestions beeps every 5 mins

lopusan-jozef, 2025-01-19:
Reporting bugs
--------------

after upgrade to AAPS v3.3.1.2 on my daughter's phone, the phone is always beeping (every 5 minutes), while she is getting suggestions for carbs even if I switch off the local alarm for carbs (tried switch ON/OFF/ON/OFF, but it does not help). It is annoying during the day, but mostly during the night, while we have added carbs, but till the BG has not started rising, it was beeping every 5mins. Thus one problem is the alarm switch OFF is not working, and second - isn't it possible to implement there some "snooze" for X minutes (configurable) for carbs suggestions?

lopusan-jozef, 2025-01-19:
[AndroidAPS._2025-01-18_12-21-27_.4.zip](https://github.com/user-attachments/files/18467972/AndroidAPS._2025-01-18_12-21-27_.4.zip)

Gohtraw, 2025-01-19:
> It was around 22:30-23:00
> 
> [AndroidAPS._2025-01-18_12-21-27_.3.zip](https://github.com/user-attachments/files/18467973/AndroidAPS._2025-01-18_12-21-27_.3.zip) [AndroidAPS._2025-01-18_12-21-27_.4.zip](https://github.com/user-attachments/files/18467972/AndroidAPS._2025-01-18_12-21-27_.4.zip)

I have the option to ignore that alarm (that way it will go off again in 5 min) or snooze it for up to 30 min. 
I think snooze is not the easiest menu item to reach because to snooze it I have to press a few buttons on my watch (OK, down, down, down, OK).

I can't download your logs - what is "Minimum Carbs Required For Suggesstion" set to? Upping that value might help.

lopusan-jozef, 2025-01-19:
I think in the older version of AAPS it was possible to set "needed carbs for suggestion" higher than 10g, now it is 10g max. This option does not help.

[image]

[image]

Regarding snooze, I meant in the mobile app, not on wearables.