nightscout/AndroidAPS #3775: [Minor Visual Bug] Automation Location - Lat & Long not correctly displayed

pnharris02, 2025-01-23:
When configuring an automation based on location, the field is not wide enough to display the number, with the default number of 6 decimal places.

The first 1-2 digits of the integer value are obscured behind the [-] sign.
The last 1-2 digits of the decimal value are obscured behind the [+] sign.

I would also argue that the default precision of display to 6 decimal places is far too accurate for any smartphone GPS.  You could, of course, just ignore the last digit (which represents a distance of only 0.11m - and yes, I absolutely had to look that up!).  But renders the + and - buttons a bit pointless, since they are only changing the value by 0.11m at a time, and you can't see the change occurring (not that I'd ever use the + or - buttons in this instance anyway!

While I'm nit-picking little things on this screen (sorry!) I would also suggest:

The screen would make more sense (to me) and be more consistent with other automation screens, if the operator drop-down box (When you are inside the area... etc.) was near the top, between the Name: and Lat: fields.

The Lon: title would be better written as Long:

Thanks for considering.
P.S. I love the automations now that I've finally made some time to start investigating them :-)  Great work!

[image]

koelewij, 2025-01-29:
I agree that the +- buttons are not very useful if there is no visual (map) reference to see where you are going anyway. 
To my opinion, people add a location with the "Current Location" button, or they copy paste values from elsewhere. 

It would be great to have a map widget under it showing when the location is, and even better, you would be able to pick a location on a map, but considering the number of times you are using this, it might not justify the effort to do so. 

But I would strongly advice to remove the +- buttons.

pnharris02, 2025-01-29:
When I looked at the code, I found it easy enough to change the precision to only 5 decimals and the +/- step to 1.1m instead of 0.11m.
But it appears (to me, as a very inexperienced coder) that the same size and type of dialogue was used for every number field, so I could not find an easy way to widen the field.
I agree that the +- buttons could be removed, and that would take away the need to widen the field. I'll see if I can work out how to do that.
I also agree that the map might be useful, but would not be used enough to justify the effort.

MilosKozak, 2025-02-25:
https://github.com/nightscout/AndroidAPS/commit/062324f6e14c934d4b11b1d46c7a1e3b6621ebe8

closed 2025-02-25 by MilosKozak