nightscout/AndroidAPS #3777: [Feature Request] Automation - Add simple maths operators to numeric condition fields

pnharris02, 2025-01-23:
I'd like to suggest the benefit of adding simple maths operators to numeric condition fields in the automation rules, where two different conditions could be combined to give a numeric result.

One usage example would be to add **Glucose** and **BG Difference**.
So, **Glucose + BG Difference < 5.0** (delta in this situation would usually be negative, hence why I've used the + operator)
This could allow customised notifications alerts, not based solely on blood glucose value OR rate of change, (which is already possible, both in BYODA and AAPS automation), but by considering both and pre-warning of the likelihood of approaching a particular value in the next 5-15 minutes.
I realise, of course, that the rate of change is not necessarily going to be the same, but I'd still like to be given this warning.

MilosKozak, 2025-01-23:
this would mean too complicated UI on phone

pnharris02, 2025-01-23:
Possibly, yes.  I'll have another think about how this might work and let you know if I come up with anything that isn't too complicated.

pnharris02, 2025-01-23:
Hey @MilosKozak 
Thanks for allowing this to be discussed. I've had a think about this, and come up with 3 options.  The first one is the simplest, and therefore possibly the best? I'll put them in separate replies, so that the screenshot mock-ups can match the text easily

1) Provide a trigger for Glucose plus Delta.
It only really covers the usage example that I've suggested, rather than allowing users to build their own expressions, but this is really the only practical use situation I can think of where I'd want to add fields.
I don't think it complicates the UI, other than adding one extra condition to the conditions list.
The subsequent trigger dialog would then work exactly the same as the existing BG difference dialog.

It would look roughly like this.  
Note that I have renamed the existing condition "BG difference" to "BG delta" because I think it makes it clearer, but that is not required if you disagree

[image]
[image]

pnharris02, 2025-01-23:
Side Note:
A completely different way of achieving the early hypo (or hyper) warning that I'm trying to achieve in the above scenario, would be to give access to the algorithm's prediction line data in the automations.
I'm just thinking out loud here, and I'm not going to bother creating screenshot mock-ups for this one, because I think it's more complicated to program and to use, but I'll mention it anyway, in case others can think of other potential uses for this
1) Make a trigger condition called BG Prediction
2) Within the trigger dialog, have
    a) a drop-down field for the time in advance that you want to look at (eg. 15min prediction)
    b) a field where you would enter the value of the glucose 
    c) a second drop-down where you chose which of the 4 prediction lines you wanted to use as well as options for MIN and MAX, which would be whatever was the lowest (or highest) prediction in the algorithm at the time-frame (15 min) chosen.

Like I said.  This one is I think too complicated.  
I'll come back soon with my other alternatives for maths operations on numeric fields

pnharris02, 2025-01-23:
The more I think about it, the other options that I was going to suggest are probably not worthwhile, UNLESS I or someone else thinks of any other practical usage situation where they would want to do arithmetic on fields. So I'll mention them only briefly.
Read ahead if you're interested, but I'd prefer that you consider my first option with screenshots above.

In fact, it may be better if we close this issue and I just create a new one with a different title for that specific usage?

Option 2: Create an option on the bottom of the trigger conditions list called "Expression Builder" or something similar.
In the trigger dialogue that opened, it could be as simple or complex as you wanted, and would only be seen by those advanced users who chose to use it.  In its simplest form it could have 3 drop-down dialogues across the top: Field 1, Operator, Field 2.
Field 1 would include those conditions that had numeric values, Operator would list the maths operator ( + or - ) and Field 2 would either list the same numeric conditions again as Field 1 (or restrict it to those that were relevant to Field 1). Then below that you would have the normal trigger components: (Equal, >, <, etc) and the numeric value for the expression result.  But this would be complicated if Field1 or Field2 had optional components (like BG delta, steps, etc), so I can't justify recommending this.

Option 3: instead of adding another condition for the trigger list, create an options button to bring in another field value within the trigger condition dialogue for numeric conditions. But I think that has too much potential to create confusion, with users thinking "why would I want to do that?".  I'm not going to bother doing a mock-up for that one.

closed 2026-10-11 by MilosKozak