nightscout/AndroidAPS #3783: Widget Background Color

gianlucaleuthen, 2025-01-25:
Please implement to change Background colors for AAPSClient widget or (at least) revert back to black Background instead of the horrible current background colors (yellow/blue). Additionally I don't quite understand the different colors anyway. Why should these be different?

MilosKozak, 2025-01-27:
you can always set opacity to zero or close to zero
different color is for parents using 2 clients to be able distinguish between them

gianlucaleuthen, 2025-01-27:
Hi Milos,

I know that opacity can bei changed, which (in my case, for example) doesn't make it any easier to read. 
What is the point of differentiating between users in fixed colours? As a user, I know which mobile phone I am currently using.

olorinmaia, 2025-01-27:
It's basically so it will be easier for parents following multiple children and to easily see which one by color. 

What about adding an option for AAPSClient 1 and 2 to toggle background color for these two areas? Standard could be like it was before and add option to change color as implemented now for those who wish to distinguish?

gianlucaleuthen, 2025-01-27:
@olorinmaia : Sounds like a good idea/compromise

ikniwn, 2025-02-08:
Agree.  Last version of AAPS had black widget background.  New version is yellow, which doesn't work as well for me.  An option to configure widget colour would be useful.

MilosKozak, 2025-02-08:
https://github.com/nightscout/AndroidAPS/commit/52b304882e23414f2f4e4742f69050cfbba6aa4c

closed 2025-02-08 by MilosKozak