nightscout/AndroidAPS #3790: [Feature Request] Separate pump connection settings from other settings during import

koelewij, 2025-01-29:
Problem:
Currently during export and import of settings all possible settings are transferred. 
For most users this is very useful, but for a lot of patch pump users, this means that pump settings will be overwritten also. 
Practically that means hat settings older than 3 or 4 days become useless because the user will lose the connection to the current pump. 
Since it can be useful to experiment with different settings/profiles etc. the requirement for reading in older settings files is an important feature.

Proposal:
During import allow the user to select if pump settings (when applicable) and/or other settings/profiles should be imported by means of tick boxes, that would be toggled on by default. Possibly this could be merged with the password dialog, so no extra steps are introduced. 
In my opinion there could be a separation between:

- [x] Preferences
- [x] Profiles
- [x] Pump connection settings

The ability to exclude profiles, could be useful if a users don't want those to be overwritten. 
I personally don't think this would have any impact on exporting of settings since all the filtering can be done on import side.

Looking forward to opinions on this.

Philoul, 2025-01-29:
Godd idea, even if it can be diffilcult to find a simple technical solution (each pump driver has it's own settings...). 🤔

vanelsberg, 2025-01-29:
Great idea. Would value such feature very much. 

As @Philoul remarks: solution would not be trivial. But I think it should be doable?

Gohtraw, 2025-04-14:
Lovely idea.
Please add a checkbox for Nightscout / Tidepool, if possible.
Reasoning: I'd like to use my configuration on virtual phones / on my dev test phone without the fear of it messing up my nightscout database (every time ~~in the rare case~~ I forget to disable all internet connectivity before I import the settings).