nightscout/AndroidAPS #3792: DynamicIFS Bootstrap

Thomas-git, 2025-01-29:
This is more of a suggestion to be discussed than a bug report.
First, many thanks to all dev. My 10 yo daughter is now able to (nearly) close loop with freestyle libre2  and DynIFS. (After many years on AndroidAPS).

# DynISF Adjust Factor
The biggest challenge was to set DynamicIFS factor right. But today, after viewing Chris Wilson video (Loop and Learn) I realise I could have set it right in the first place. Here is how:

DynIFS formula is:
ISF = 1800 / ((TDD * DynISF Adjust Factor) * Ln (( current BG / insulin divisor) + 1 ))

Wilson says 1800 rule equal this formula at 150 BG, so we have:
DynISF Adjust Factor = (1800 / ISF) x (1 / TDD) x (1 /  Ln (( 150 / insulin divisor) + 1 ))

TDD can be extracted from user data (or also bootstraped by user, see below)
ISF should be the lowest ISF at night in user profile or maybe the lowest overall. This is because TDD will increase while bolusing during the day and so will DynamicISF.

For my daughter it gives 43% factor whereas my 2 weeks trial and error gave me 40%. So that’s pretty close.
Please consider adding a "auto" button next to DynISF Adjust Factor in settings to help every body set it right.

# TDD
Next thing to consider is a possibility to adjust TDD calculation (seen on another issue but I can’t find it right now). My daughter (maybe all children?) need a quicker changing TDD calculation to follow quick ISF swings. The revised formula (seen elsewhere and capped to 5 days) is:

`val tddWeightedFromLast6H = ((1.1 * tddStatus.tddLast3H) + (0.9 * tddStatus.tddLast6to3H)) * 4
            dynIsfResult.tdd = ((tddWeightedFromLast6H * 0.6) + (tddStatus.tdd5D * 0.15) + (tddStatus.tdd1D * 0.25)) * preferences.get(IntKey.ApsDynIsfAdjustmentFactor) / 100.0 * multiplier`

Maybe a slider allowing to choose between fast changing ISF or steady ISF would be great. Or maybe original formula it good for adults and this one for children?

# Overall bootstrap
Another idea is to help bootstap AndroidAPS and it’s most advanced algorithm per CamAPS idea: Ask user 5 days average TDD and body weight to figure out a simplified profile. Instead of waiting a week to enable DynIFS, it can use a fallback average TDD in profile. Body weight would help with TDD calculation ratios (small weight = less blood = more swings). Base ISF would be computed from 1800 rule. Finally basal rate can use data from [studies](https://pubmed.ncbi.nlm.nih.gov/21545675/).

szantos, 2025-02-04:
Please modify the expression "revised formula" to anything else not suggesting that it is something newer or better. It was only an iteration of my custom (n=1) tuning of the formula and should not be generalized. Feel free to show it as one way of making DynISF more responsive, but it is by no means an update or a development of the original formula. (Maybe put "dynIsfResult.tdd = ..." in a new line to enhance readibility.)

Thomas-git, 2025-02-04:
You’re right. n=1 experiment should not be generalized.