nightscout/AndroidAPS #3793: SMB specific settings not working

32423676573, 2025-01-29:
AAPS 3.3.1.3 - also seen going back to 3.3.1.0

SMB specific config settings are not being respected. I have SMB configured for with carbs and 6 hours after carbs. I am receiving carbs all throughout the day. I also have no carbs for high temp and am receiving then with high temp targets as well.

I have not tested every configuration option but can do some further testing. Let me know what you need to help troubleshoot.

Screenshots and treatments below. You can see if continued to receive carbs throughout the night. The last carbs were at 7:45PM so 6 hours would have put that until 1:45 but I have carbs up until 5:52.

[image]

[image]

[image]

[image]

[image]

vanelsberg, 2025-01-29:
> I have not tested every configuration option but can do some further testing. Let me know what you need to help troubleshoot.

I's start switching of the option "Enable SMB after carbs".
You should then only see SMB active when COB > 0

32423676573, 2025-01-29:
This was the configuration I had prior to 3.3.1.0 and it worked correctly. 

Right now it's functioning as though I had SMB always enabled.

I am thinking it is related to merging DynISF directly in rather than leaving it as a plugin.

It's possible the behavior is a result of having the combination of DynISF plus the SMB settings enabled.

I rely heavily on DynISF. I can try turning it off overnight and see if it stops giving SMB all the time.

reijoahola-nightscout, 2025-01-29:
The screenshots do not show at what time carbs has decayed away (Cob=0).

32423676573, 2025-01-29:
60 carbs at 7:45PM would not take 12 hours to decay.

Here's a screenshot showing there are no carbs overnight but SMBs are delivered.

[image]

reijoahola-nightscout, 2025-01-29:
Note that "Enable 6h after carbs ...".

32423676573, 2025-01-29:
Exactly. Thats much longer than 6 hours after carbs. This is the same settings I used pre 3.3.1.0.

The behavior changed.

MilosKozak, 2025-02-23:
could you paste a screensot from SMB tab when this happen?
there are applied contraints listed in top section

MilosKozak, 2025-09-25:
closed for inactivity

closed 2025-09-25 by MilosKozak