nightscout/AndroidAPS #3796: Unable to Install or Upgrade AAPSClient on Android 14 (Samsung Galaxy S23 Ultra)

yoeldk, 2025-01-30:
I  am unable to install or upgrade AAPSClient on my Samsung Galaxy S23 Ultra (Android 14). I have tried both upgrading from an existing installation and performing a fresh install after clearing cache and data, but neither works.

Steps to Reproduce:

1. Attempt to install the latest version of AAPSClient 3.3.1.3

2. If upgrading, I get the error:
"App not installed as package conflicts with an existing package." (See attachment)

3. If performing a fresh install:
First, I receive a Google Play Protect warning stating that the app was built for an older Android version. (See attachment).

After selecting "Install anyway", I get the same package conflict error.

What I've Tried:

Uninstalling AAPSClient before installing the new version.

Clearing cache and data before uninstalling.

Restarting the device after uninstalling before attempting reinstallation.

[image]
[image]

emilisev, 2025-01-30:
Do you also have AAPS installed on this phone ? If yes, you need to compile AAPSClient yourself, because you can't have both compiled with different keystore.

emilisev, 2025-01-30:
Yes that's a new restriction starting from this version

yoeldk, 2025-01-30:
I'm sorry, but I don't understand the logic behind this change. Why should my personal AAPS be linked to AAPSClient, which is solely used to monitor my daughter's sugar levels?

emilisev, 2025-01-30:
Check the doc if you don't believe me 😉 : https://androidaps.readthedocs.io/en/latest/RemoteFeatures/RemoteControl.html#download-and-installation

I believe it is not a decision from aaps team but some limitation induced but the android libraries used (they were updated in last version).

yoeldk, 2025-01-30:
I believe you, but I just don't understand the logic behind this change, especially since it works perfectly in version 3.2.0.4. If this change was imposed by Android libraries rather than the AAPSClient developers, then it's a different matter and beyond our control. I just want to make sure that this is indeed the case.

emilisev, 2025-01-30:
I guess it has something to do with 2 aaps from the same author but different keystore, since the keystore is used to authenticate the app, but I'm no specialist

yoeldk, 2025-01-30:
The question is what has changed between AAPSClient 3.2.0.4 (everything works fine) vs 3.3 and up (won't install due to conflict).

emilisev, 2025-01-30:
Aaps uses a lot of android librairies, and a lot of them were updated (hence the requirement for Android 11 as well), I guess that's where it comes from.

MilosKozak, 2025-02-23:
it's android limitation. Apps that use the same permission must be signed with the same key. Permission is new in 3.3 to allow sending data to watch from multiple sources

closed 2025-02-23 by MilosKozak

yoeldk, 2025-04-13:
@MilosKozak so just to clarify, does this mean that any diabetic parent who use AAPS for himself and aapsclient for one of his diabetic kids, needs to build the aapsclient by himself instead of downloading it from here? I'm pretty sure there are many such cases of diabetic parent + diabetic kid.

yoeldk, 2026-01-06:
Well, I didn't build AndroidAPS myself, and the client APK was downloaded from here, so now I need to build both projects myself instead of none.  Ok, if there's no other choice, then that's what I'll do... Thanks!

MilosKozak, 2026-01-06:
where was your main apk from?

yoeldk, 2026-01-06:
I got it from a friend