nightscout/AndroidAPS #3797: [Bug 3.3.1.3] AAPSclient does not fetch data from NS

emilisev, 2025-01-30:
Happens really frequently (several times a day) : suddenly AAPSClient stops receiving data from NS. If I hit "deliver now", I receive the data at once. I have tried with or without websockets with same result.

It started happening on 3.3.1.3 (my previous version was 3.3.1.0). My phone has proper internet access, because every time I see the last value displayed correctly on xdrip.

[AndroidAPS.log](https://github.com/user-attachments/files/18606533/AndroidAPS.log)

It can be seen in this log several time during the day, from : 
08:03 to 08:21, 
08:49 to 09:04,
09:32 to 09:51,
14:17 to 14:35,
14:38 to 15:02,
15:03 to 15:47,
15:48 to 16:30, 
16:59 to 17:15,
17,43 to 17:54.

olorinmaia, 2025-01-30:
I'm guessing you are using NSClientv3? 

Could you try on your AAPSClient, go to maintenance and reset DB. Exit app. And then do a full sync. And see if you experience it again?

emilisev, 2025-01-30:
Yes NS v3. Thanks for the suggestion, I've tried that and will update here.

emilisev, 2025-01-31:
Unfortunately it didn't solve the issue.

fedemotta, 2025-02-27:
I am with the same issue. Same version.

emilisev, 2025-03-02:
The issue was solved on my end after a factory reset of my phone (for another reason, I didn't expect it would solve this).
I suspect that it could be due to the synchronization to the time server disabled on my phone, and being a few minutes off, but I have not verified that.

fedemotta, 2025-03-02:
I fixed deactivating nightscout cloud upload in the xdrip+ application.

zreptil, 2025-03-26:
Same here. When using NSClientV3 the glucose data from nightscout is not received. When switching back to NSClientV1 then it works again. I updated to 3.3.1.3 a few hours ago and used 3.3.1.0 for several weeks where the same occured. I tried to switch to NSV3 since the battery drain of NSV1 seems to be very high. At least i could not figure out any other app that could be responsible for this battery draining.

What I have done so far to make NSV3 work:
- close AAPS completely and restart it
- clear the database in maintenance
- reset the database in maintenance
- checked time settings in phone settings (is set to automatic)
- reboot the phone

the phone is a pixel 6 from google

MilosKozak, 2025-03-26:
If you don't receive data using v3 it only does mean it in not uploaded using v3

zreptil, 2025-03-26:
Perfect, thank you so much :) I just had to update Juggluco to send it using v3 (declared as test in the app, but working) and now NSV3 works. Great 👍

MilosKozak, 2025-03-26:
Still not best practice
Proper way is Jugguco -> AAPS > NS

zreptil, 2025-03-26:
I looked for that possibility, but could not find it. So I stuck with reading the glucose values from nightscout. What have I to configure in aaps to read the glucose values from juggluco?