nightscout/AndroidAPS #3798: [Bug] - Bolus Wizard does not correctly display negative units

pnharris02, 2025-01-31:
The Bolus Wizard/Calculator now shows a positive number for IOB when you have positive IOB. In release 3.2.0.4, it was a negative number, which was logical because it had to be deducted from the calculated dose.

This was brought up in Discord by @barisk81 and discussed by myself and @robby
https://discord.com/channels/629952586895851530/1199783453793734677/1334593953734983743

I suggest, to reduce confusion, the actual IOB is shown in the horiz centre of the dialog, above the COB figures, then the inverse/negative number is shown in the right column, similar to v3.2

A second visual bug on this screen is that any negative number in the right column causes the "U" after the figure to wrap to the next line.  And in the case of IOB, the bottom of the U is also truncated slightly.

The attached screenshot pairs are from two different times, one in September to show 3.2 behaviour and one from this morning on 3.3.1.3, with a rough sketch of my suggested resolution

[image]

[image]

[image]

[image]

robertrub, 2025-01-31:
Yes, it would be logical to keep the same math notation for all the numbers. If it is added to the bolus calculation, it should be + (or no sign) but if it will be subtracted, it should have the - sign in front.

olorinmaia, 2025-02-17:
Fix https://github.com/nightscout/AndroidAPS/pull/3831

Is merged to dev for those who want to test

pnharris02, 2025-02-17:
Thanks @olorinmaia. I'll download the latest dev and check it out when I get a chance.
I'm inferring you didn't like my idea of displaying the actual IOB (the inverse of what is shown in the calculations column) in the middle column of the page?
But that's okay if you've changed the sign. Thank you for fixing this.

olorinmaia, 2025-02-17:
@pnharris02 Your welcome :) I focused on getting correct math notation in wizard calc log so when you sum them all up it will be the bolus you will administer in the end. I think it's sufficient to show the IOB amount at the right when you use bolus calculator. It's not a bad idea, but there is always something about adding more text to the screen, it need to be really necessary imo if added.

If you go to the Treatments tab and open "Calc"-log, you can see both bolus IOB and basal IOB also. If you think this fixes the main feel free to close this issue :)

pnharris02, 2025-02-17:
Thanks @olorinmaia
I'm not home at the moment but will test as soon as I can

pnharris02, 2025-02-18:
@olorinmaia  I am not familiar enough with the dev version/process to try this on my live device, but I have reviewed your code changes (with my very limited coding knowledge!) and have run the dev version using an emulator of my device in Android Studio, and agree it has added the negative sign correctly to the bolus wizard calculations.

But the other part of my bug report was that in the latest AAPS version 3.3, my device is wrapping the "U" to the next line.
I note that this behaviour is not seen on your before or after screenshots in https://github.com/nightscout/AndroidAPS/pull/3831
I also note that the behaviour is NOT seen in the Android studio emulation of my device, even with the old code.
So perhaps this is something specific to my device, which is a Google Pixel 9 Pro running Android 14.
It was definitely not there back in September, but at that time I was running a different device.  I changed devices in early December, and then upgraded from AAPS 3.2 to 3.3 in late December.  But I don't have any screenshots from mid December that would allow me to comment on whether or not the problem existed in 3.2 on my new device.

I'm not sure if this is something you can fix if you're not seeing it on your device or in emulations.  Is there any other information I can give you to help to track down the source of the problem?

olorinmaia, 2025-02-18:
I noticed your problem with the U getting wrapped out but not able to reproduce it. I've done some other small modifications to UI checkbox also that possibly could fix it but it's a long shot.

I tried enabling superbous or more or less zoom but it didn't help. Is there perhaps any special setting you got activated on your end?

pnharris02, 2025-02-18:
No special settings that I can think of that would affect that, but I wouldn't really know what could cause that behaviour. 
If you have any hints, then I can try different things.  

When I have some time, I will try to install 3.3.1.3 on an old Android 9 phone and Android 11 tablet and see if it occurs on either of those.
I guess I could try to see if I can create another emulation of my phone in Android Studio with Android 14, and see if that replicates the problem, but there is of course no guarantee that an emulator would exactly replicate the problem even after doing this.

I could also try upgrading my phone to Android 15, but I've been wary of that because I'm an Omnipod user.
And the "U" wrapping around a line is of course a very small problem to want to take too many risks looking for it!

pnharris02, 2025-02-19:
I configured an emulator in Android Studio for Google Pixel 9 Pro with Android 14 (14.0 Google APIs) and it does not show the problem that I'm seeing with my real phone.  So no closer to being able to identify/replicate the problem yet.
If I compare the emulation with my real phone, everything looks the same, except that:
- On my real phone, the U wraps to the next line, and the number right aligns to the very edge of the screen. 
This also causes a change in the vertical alignment of the the values in the middle column
- On my real phone, the text is relatively slightly larger, such that the "15min trend" pushes right up against the trend value beside it

[image]

[image]

robertrub, 2025-02-21:
`On my real phone, the U wraps to the next line, and the number right aligns to the very edge of the screen.`

The character sizes are different in your 2 screenshots. Try to reduce the character size on your phone.

[image]

pnharris02, 2025-02-21:
Yes, I noticed the character size difference too, and commented on it above.
I don't think I changed the font size when I got the phone only 2 months ago.
But yes, if I change it to a smaller font, the U does fit on the same line
I think I prefer the original font size though :-)

olorinmaia, 2025-02-23:
@pnharris02 Since the main issue with wrong math notation has been fixed in dev, would you consider closing this issue, and perhaps consider opening another issue for your font size problem and steps on how to reproduce that issue?

pnharris02, 2025-02-23:
@olorinmaia Sure.  Will do

closed 2025-02-23 by pnharris02