nightscout/AndroidAPS #3801: Spammy User Entry Care portal (CGM Sensor insert)

piecycle, 2025-02-01:
Since a couple of releases, the user entry tab in the Treatments section logs the same message every 5 minutes (see screenshot).

[image]

I'm using a Dexcom G7 with xDrip as collector and data source in AAPS. The screenshot was taken from AAPS 3.3.1.3 (latest Dev from today, 01/02/2025), running the tsunami mod, but the issue appears to be unrelated to the mod.

olorinmaia, 2025-02-01:
Also see https://github.com/NightscoutFoundation/xDrip/discussions/3508 for possible solutions. Something has to be done, either on xDrip side or AAPS side, there is nothing decided yet.

reijoahola-nightscout, 2025-02-02:
I checked one earlier CGM Sensort Spam list. I am sure that earlier it was quite small, may ne 10 lines. Now it was over 110 lines ! Spamming seems to continue and increase the list.

vanelsberg, 2025-02-03:
> logs the same message every 5 minutes 

Current dev, using BYODA; not seeing this. So I'd say this is a xDrip+ problem only.

olorinmaia, 2025-02-03:
If using xDrip+ and G7 and experiencing this problem, please see https://github.com/nightscout/AndroidAPS/pull/3804 and test if you want. It fixed the issue for me. I haven't changed G7 session yet with this fix so that is still to confirm.

olorinmaia, 2025-02-05:
Fix is now merged to dev branch

olorinmaia, 2025-02-10:
Can confirm that new sensor start is received when changing G7. Still no spam. @MilosKozak I think this issue can be closed then.

olorinmaia, 2025-02-23:
@piecycle @MilosKozak This issue can be closed due to ce20cccf - tested OK on several sensor changes.

closed 2025-02-23 by MilosKozak