nightscout/AndroidAPS #3806: [Wear OS] Bad display of negative IOB

keweki, 2025-02-04:
The negative symbol is not display on Wear OS complications.
[image]

[image]

olorinmaia, 2025-02-04:
Strange. This is working on my watch, both dev and master. Just checked all the IOB complications now.

[image]

keweki, 2025-02-04:
Thanks for your check. I tried the other complications too and I have the same issue 😔
What is your watch model please ?
Mine is Galaxy Watch 7 44mm

olorinmaia, 2025-02-04:
Galaxy Watch 4 40mm attached to test-rig! I also got a GW 5 44mm attached with master-rig, can check next time it got negative IOB. :) Could there be some language/regional setting on your phone/watch causing this? I'm using Norwegian or English language / region.

olorinmaia, 2025-02-04:
So when I switched to English language on phone, and in AAPS, after a while I experience same issue on my test-rig. So it's related some how. I notice in norwegian format is "-0,00". but in english "-0.00".

olorinmaia, 2025-02-05:
Could you change this somehow to make it work? I wonder how code could be adapted to make it work for English 🤔

keweki, 2025-02-05:
I will try next time my IOB will be negative

MilosKozak, 2025-09-25:
closed for inactivity

closed 2025-09-25 by MilosKozak