nightscout/AndroidAPS #3817: Bolus and calc log doesn't get synced to AAPSClient from AAPS when using bolus calc occasionally (NSCv3)

olorinmaia, 2025-02-09:
AAPS / AAPSClient 3.3.1.3
Build: 2a10ec79a6-2025.02.07
Nightscout-versjon: 15.0.2
NSClientv3

Occasionally the bolus and calc log from bolus calc doesn't get synced to AAPSClient. It's uploaded to NS. This time we used bolus calc on watch. Look for bolus calc entry and bolus 15:41

It's the third time happening since upgrading to 3.3.1.3. Attaching logs and screenshots.
Workaround is to clear database in AAPSClient and do a full sync.

Logs:
AAPS
[AndroidAPS_LOG_1739112889790.log.zip](https://github.com/user-attachments/files/18725045/AndroidAPS_LOG_1739112889790.log.zip)

AAPSClient 
[AndroidAPS_LOG_1739112935928.log.zip](https://github.com/user-attachments/files/18725046/AndroidAPS_LOG_1739112935928.log.zip)

Screenshots:
AAPS
[image]

AAPSClient
[image]

NS
[image]

MilosKozak, 2025-02-10:
AAPS log from mentioned time is missing

olorinmaia, 2025-03-03:
@MilosKozak Providing another example with logs as it happend again now. Look for  bolus calc treatment at 17:18.

Logs
AAPS: [AndroidAPS_LOG_1741019076772.log.zip](https://github.com/user-attachments/files/19057060/AndroidAPS_LOG_1741019076772.log.zip)
AAPSClient: [AndroidAPS_LOG_1741019040811.log.zip](https://github.com/user-attachments/files/19057066/AndroidAPS_LOG_1741019040811.log.zip)

Screenshots:
| AAPS | AAPSClient |
|--------|--------|
| [image]| [image]|

olorinmaia, 2025-08-27:
Haven't happend in several months. closing issue