nightscout/AndroidAPS #3819: Android 3.3.0 and upwards, huge increase in battery drain

SimonPhilpott, 2025-02-09:
Reporting bugs
--------------
This is observed in all major releases after AAPS 3.2.0.4 including the latest 3.3.1.2 Build: 202df2cebf-2025.01.15

Battery drain was previously around 5% per day. 

This has jumped to around 20%. 

[AndroidAPS._2025-01-05_00-00-21_.1.zip](https://github.com/user-attachments/files/18726058/AndroidAPS._2025-01-05_00-00-21_.1.zip)

[image]

I've tried a fresh install rather than update, and clearing cache. 

Rebooting obviously doesn't work. I'm using the NSClientV3 but have also tried disabling uploading to eliminate this as a possible culprit. 

Setting are completely unchanged between 3.2.0.4 and 3.3.1.2, hardware is also completely unchanged. 

Re-installing 3.2.0.4 lowers the daily battery drain back down to 5%, but this will soon be an unsupported version so this isn't a permanent fix.

MilosKozak, 2025-02-10:
reset logging to default

SimonPhilpott, 2025-02-10:
Hi thanks for the reply,

Just to check. Do you mean the files found at 

/storage/emulated/0/AAPS/logs/info.nightscout.androidaps?

jaredwhittle, 2025-02-10:
I think he means first, go to Maintenance > Log Settings > (scroll to the bottom) Reset to Defaults button
Then for remove old db records, go to Maintenance > Database Cleanup > OK to removed tracked changes and historic data older than 3 months

SimonPhilpott, 2025-02-10:
Is 6h of wakelocks normal?

[image]