nightscout/AndroidAPS #3825: Regarding the graph text bug from v3.3

MiyeongKim77, 2025-02-12:
Starting from v3.3, for Dana R pump users, when the basal increases in the aapsclient app, the extended bolus is output as text instead of the increased basal graph. The problem is that these texts overlap and are not visible, so we would like it to be changed so that the increased amount is displayed as a graph like before. It is not displayed like that in other pumps.. The reason it is displayed like that only in Dana R pumps is probably because Dana R pumps use extended bolus when basal increases or decreases.. Since there are users in Korea who use AAPS with Dana R pumps, we would appreciate it if you could let me know which developer I should contact to fix this.

[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/GettingHelp/AccessingLogFiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

1BeuMi1, 2025-02-28:
Hi, the AccuChekinsight Pump Shows exactly the Same behavior! There are the Same Display Problems. The pump also use extended bolus when basal increases or decreases. The behavior did not exist in the previous Versions. And for me this only occcurs with the AAPS Client.

MilosKozak, 2025-03-01:
fixed by https://github.com/nightscout/AndroidAPS/commit/4e71ae0d29df5b8c214b92f9cd3fba0a3040ebb3

closed 2025-03-01 by MilosKozak

MiyeongKim77, 2025-03-01:
Thank you so much. Kozak.

1BeuMi1, 2025-03-01:
Thanks a lot Milos!

1BeuMi1, 2025-03-10:
@MilosKozak  When will the update for the AAPs client be available for download?

1BeuMi1, 2025-03-10:
Thanks for quick response 😊 When will the release come approximately?