nightscout/AndroidAPS #3830: Equil Pump IOB / History incorrect on battery failure / stop

Stealthwick, 2025-02-14:
Currently testing Equil pump with AAPS 3.3.1.3

I've come across an issue where AAPS did not sync pump history correctly when there was a gap in insulin delivery of several hours due to the pump battery running flat.

After changing the reservoir and battery and re-connecting the pump, the delivery history still showed profile basal delivery for the period that the pump was inactive, with incorrect IOB as a result.

@EquilHack logs available if you need

MilosKozak, 2026-10-11:
Part of this is fixed on dev in 2dbdd5fd83: **a suspended or stopped pump now gives 0 basal IOB**.

Until now, when the Equil went to SUSPEND or STOP (from the Equil screen, from the pump, or during a reservoir change), AAPS only stored the running mode. IOB is calculated from temp basals, so it kept counting basal that was never given.

Now:
- RUN → SUSPEND/STOP records a 0 U/h `PUMP_SUSPEND` temp basal. The running loop temp basal ends at that moment.
- SUSPEND/STOP → RUN ends that record.
- After a resume the pump continues the temp basal it had before the suspend (the pump history shows rate 0 after resume for an old 0 U/h temp basal). AAPS has already ended that temp basal, so it is now cancelled on the pump after the resume. The loop sets a new one on its next run.

Tested on a real Equil: suspend/resume from the Equil screen and a full reservoir change (STOP → RUN). The records are written, ended and uploaded to Nightscout.

Also fixed in 37daa86fe2: after an app restart the queue could wait 2 minutes before it connected to the pump, because a stored "connecting" state was loaded.

**Still open:** the case in this issue, a battery that dies while the pump runs. The pump is not in SUSPEND/STOP then, so this fix does not cover it. Leaving this open for that part.

MilosKozak, 2026-10-11:
The battery part is fixed on dev too, in 7125532a9e.

When the battery dies, the pump logs "Battery empty!" with its own time and stops delivering in the same second. This was checked on the history of a test pump whose battery really died (2026-08-03). AAPS reads that record when the pump has power again, and now records 0 U/h from that exact moment until the pump shows delivery again (or until the next RUN). Because the start is the pump's own time, the gap is never made longer than it really was.

Together with the suspend fix from the comment above, IOB no longer counts basal that the Equil did not give. The same mechanism now also covers an empty reservoir (#3467). Closing.

closed 2026-10-11 by MilosKozak