nightscout/AndroidAPS #3836: [Wear] Display/Rounding error when setting preset temp target from AAPS Wear

pnharris02, 2025-02-18:
There appears to be a rounding/display error when setting a preset temp target from the watch.
It only appears with some mmol values (eg. 9.8 and 9.6 but not the other numbers surrounding this) so I wonder if perhaps there is some sort of rounding bug when/if it is converted to mg/dL and back again??
See screenshot

[image]

Also, while you're there, it would probably be good if the title was changed from "Temptarget:" to "Temp-Target" or something similar

olorinmaia, 2025-02-18:
My proposal to fix this, works fine on my end: https://github.com/nightscout/AndroidAPS/pull/3838

pnharris02, 2025-02-19:
Looks good to me.  Thanks @olorinmaia

Philoul, 2025-02-19:
Solution proposed is fine.
Just little comment, I prefer "Temp-Target" shorter than "Temporary Target".
Keep in mind that a lot of langages use longer words compared to english (german, french, ...), and translators will generally translate long words by long words, shorter words by shorter words, without knowing where the string is used.
Short version or acronyme (Temp-Target, TT) is well known by all users because use everywhere in AAPS app and in doc 😉.

olorinmaia, 2025-02-19:
@Philoul Good point, updated PR.

olorinmaia, 2025-02-23:
@MilosKozak This can be closed due to https://github.com/nightscout/AndroidAPS/pull/3838

closed 2025-02-23 by MilosKozak