nightscout/AndroidAPS #3839: Accu Check insight pump doesn't connect to AAPS app

vriesdef, 2025-02-19:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem

A few days ago I started AAPS for the first time with Dexcom G6 and Accu check insight. I completed all the steps and a problem occured. After finishing all the steps my pump doesn't connect with AAPS. My pump is not connecting to another device. I switched BT off and on. Today I started form scratch with another laptop and the problem still occures. I have 2 accu check insight pumps and tried both.

https://github.com/user-attachments/assets/674d013f-8a67-4a4e-a534-d27dff7f1431

I already asked for help at facebook and discord

I have a samsung A14
Android version 13

- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).

Build: dece928837-2025.02.16

- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://wiki.aaps.app/en/latest/GettingHelp/AccessingLogFiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

Philoul, 2025-02-20:
Can you please include log files to help analysis?

@MilosKozak I suspect some missing authorization, but I'm not a specialist of Bluetooth, so I have no ideas on how to fix this issue and could not reproduce it...
Any ideas?

vriesdef, 2025-02-21:
[AndroidAPS\[1\].log](https://github.com/user-attachments/files/18902526/AndroidAPS.1.log)

Philoul, 2025-02-21:
Hum, I can't find any log information used within Paring process of insight pump within your log file...
Did you made log export just after (only a few seconds) pairing trials?
If yes we can try to include additional log information within code to understand the issue...

Philoul, 2025-02-21:
Note that "PUMP" should be selected within Maintenance > Log settings (and you can disable temporary all the other parameters to limit log information to PUMP for our analysis...

vriesdef, 2025-02-21:
I deleted de app on my phone a couple of days ago to rebuild AAPS form scretch on a different computer. When I finische and I couldn't pair my phone I immediatly quit. :(

I just tried to connect again and made log file. 

[AndroidAPS\[1\].log](https://github.com/user-attachments/files/18906574/AndroidAPS.1.log)

Philoul, 2025-02-21:
Hum, the only information from Insight Driver are below, don't see anything linked to Insight Pairing trials
[image]

So we will probably have to add additional log information to capture the issue (I suspenct missing authorization but not sure...)
@MilosKozak or @TebbeUbben any ideas ?

vriesdef, 2025-02-23:
Today I tried to connect with the pump and an old samsung telephone and it worked. Do you think I can connect with my new phone A14? Or is there some setup so the accu check insight can pair with the A14?

Philoul, 2025-02-23:
Insight pump parameters and AAPS parameters are the same whatever the Android version of your phone, so you can try with your Android 14 phone.
Within your video it's strange because we don't see the pairing activity starting.
Try to reboot your phone before 🤔

vriesdef, 2025-02-25:
I did a reset with my phone. Turned it of and on for a couple of times and able to pair the pump with AAPS. Thnx for the help!

closed 2025-02-25 by MilosKozak