nightscout/AndroidAPS #3840: Move 3.3.1.3-dev version display from overview

vanelsberg, 2025-02-19:
**AndroidAPS 3.3.1.3-dev**
**Commit**: [e89704e](https://github.com/nightscout/AndroidAPS/commit/e89704eb7e54745fdce219d78fbdc1063217f2d6)

Showing version on overview only clutters the already busy screen.
Overview is about AAPS state (dynamic), version is static and already (in more detail) available on the About screen.

I vote for undoing [Commit e89704e](https://github.com/nightscout/AndroidAPS/commit/e89704eb7e54745fdce219d78fbdc1063217f2d6)
and look for alternative ways?

**Proposes solution:**
I get the intension of the original commit was to make version viewable on screenshots for better user support.
This could be done by moving "About" to an additional "About" button at the upper right-top of overview. Half the size, to the right of the "Temp target" button? It can show current version as the buttons label and make "about" more accessible.

[image]

vanelsberg, 2025-02-20:
(Open for discussion)

olorinmaia, 2025-02-20:
I think it should only show, if app version is not the latest master version. It would then encourage users to update to latest master, and it will be easier for us giving support to know what version they are on. If we don't see a version there, we will know it's the latest master.

For older versions than 3.3 it's easy to distinguish as the overview graph buttons have changed.

Else I'd vote to revert it as well. I can't think of a place this should be showing all of the time in overview.

robertrub, 2025-02-21:
Yes, _show only if old_ is a very good idea. It would be nice to tap on the version information in order to get the latest available.

ie a text _v3.2.0.4_ on graph (or button...) you tab on it and a window opens with
```Your version : 3.2.0.4
Available version : 3.3.1.3```

vanelsberg, 2025-02-21:
@olorinmaia @robertrub 

Yes, agree.
Would favor showing only if old.

olorinmaia, 2025-02-21:
See 86c56831 - Big improvement in my eyes

From my test-rig:

[image]

vanelsberg, 2025-02-22:
@MilosKozak [86c56831](https://github.com/nightscout/AndroidAPS/commit/86c56831f85c239acdc09cc239981bea48ad973a) is a lot better. For me, solution accepted!

robertrub, 2025-02-22:
@MilosKozak I find the new position ok but the color is way too agressive.  Why red, the alert color ? It is the last dev, so I'm not "late".

It should be either gray or green (as long as I have a recent version).

[image]

MilosKozak, 2025-02-22:
Red is only with uncommitted changes

robertrub, 2025-02-22:
Ok. I had changed the profile % limits to 30% and 150% for the automations. That's probably the reason for the "uncommitted" code.
I'll try to reset them back to "normal" as I don't use them anymore (testing AutoISF for the moment).

closed 2025-02-23 by MilosKozak