nightscout/AndroidAPS #3841: If the DiaKem has a BG value above 400 mg/dl (Diakem display "High") it does not transmit a BG to AAPS (AAPS does not have a BG)

TeufelHH, 2025-02-21:
AAPS 3.3.1.3 
DiaKem 1.6.1.4537
Accu-Check Insight pump
Dexcom G7

Yesterday I had a BG value over 400 the entire night, so Diakem only displayed the word "High" instead of a BG value.

At around 11:30 p.m. the BG value briefly fell (to just under 400 mg/dl) and AAPS immediately received/displayed BG values ​​again.

The entire time when the BG value is over 400 (and AAPS has no BG values) the Lopp doesn't work (no SMB / basal rate increase, only the standard basal rate runs) which can't be what you want because the BG then never goes down.

I didn't have this problem with the G6 (with BYODA), so something has to happen so that the loop still works even with a DiaKem value of "High" and tries to get the BG down with SMB/basal increase.

I don't know if the logs fit the benefit but I'll just send them along.

The problem occurred on February 20, 2025 around 10 p.m. until February 21, 2025 around 6:30 a.m

[AndroidAPS.log](https://github.com/user-attachments/files/18902521/AndroidAPS.log)

blaqone, 2025-02-21:
This is not a DiaKEM problem. 
When > 400 the sensor itself does not give back numerical values. Instead it goes to „High“. So as a result this can only be fixed in AAPS (which I doubt will never happen).

TeufelHH, 2025-02-21:
I didn't claim that it was a DiaKem problem. But something has to be done in AAPS so that AAPS (if it gets reported "High" by DiaKem) takes at least 400 mg/dl or 450 mg/dl as a "calculation basis" and at least "tries" to get the BG down. Not like now that it simply stops working and only the standard basal rate is delivered.

MilosKozak, 2025-02-23:
Log doesn's contain specified timerange
but 400 doesn't mean a BG but an error state .....

TeufelHH, 2025-02-24:
Today I installed the BYOAD version built as G7 for testing purposes. Let's see if it works or if the same problem occurs with values ​​above 400 mg/dl.

Is it perhaps possible to evaluate the string that DiaKem sends (perhaps "High" is actually sent as a string) in AAPS so that the loop continues to try to lower the BG with SMBs?

If this happens in the middle of the night and the basal rate is well tested, then the BG stays at the same level the whole time because the basal rate (standard 100%) is intended to maintain the BG and is not intended to correct it. Strangely enough, it also worked with the G6, which is why I installed BYODA today.

blaqone, 2025-02-24:
Use Juggluco!!!

closed 2026-10-15 by MilosKozak