nightscout/AndroidAPS #3842: AutoISF : No variable_ISF curve if it's alone

robertrub, 2025-02-21:
@Philoul and @ga-zelle If I set variable ISF curve with the IOB and COB graph, or with the dev graph it's visible.

[image]

[image]

But, if it's alone,  the curve is not visible. 

[image]

Or, sometimes, only some dots are visible...

[image]

If needed, the logs. Look at 5:30 AM when I set it up on an individual graph.

[image]

ga-zelle, 2025-02-21:
@Philoul I won't have time during the day. But some graphs look like the last data point of VSENS might be missing. Also, I'll try DynISF on a test phone to see, whether it behaves the same way.

Philoul, 2025-02-21:
Ok, understood the issue and I could reproduced it:
SubGraph vertical scale is given by the first selected item in menu or in item list in upper left of each graph.

If you switch V_SENS graph within another graph with a higher priority item like IOB, DEV..., then graph values are calculated again to fit the new scale.

It seems like when you switch back the V_SENS curve to be the master scale (alone or with lower priority curves like STEP or HR), then scale is correctly updated on the left of the subgraph. but the curve points are not so they remains with the value of previous graph (so with an IOB scale for ex.).

I will now analyse code (above analysis onmy managed playing within the app on my phone 😉), to fix it...

Philoul, 2025-02-22:
I made a PR to fix this issue #3844 if you want to test it...

It works fine now on my side 😉 ...

robertrub, 2025-02-22:
@Philoul Built it. Working OK. I can change the place of v_ISF and it is always visible, even when it's by itself.

ga-zelle, 2025-02-22:
Tested OK. Is very robust, I cannot break it with various combinations.

However, when starting fresh the very first VSENS value is not displayed although the scaling is already updated from NaN and shown correctly. This may be a different issue.

closed 2025-02-22 by MilosKozak