nightscout/AndroidAPS #3846: No connection to Insight with AAPS 3.3.1.3

kneuZ, 2025-02-22:
Hello,

I can no longer connect to my Accu Check Insight insulin pump with version 3.3.1.3. 
It makes no difference whether I do an update or a complete reinstallation. 

I find the pump during the pairing process, but when I select it nothing happens or the message “Connection failed” appears.

With version 3.2.0.2 the connection works without any problems. 

My smartphone is a Xiaomi Redmi Note 9 pro with Adroid 12.
All authorizations have been set and the energy-saving measures have been deactivated. 

I have attached a log file from the “Pump Control” version. 

[AndroidAPS.log](https://github.com/user-attachments/files/18922602/AndroidAPS.log)

[image]
[image]
[image]

MilosKozak, 2025-02-22:
@Philoul

Philoul, 2025-02-22:
Within pump UI
If you go to device settings, communication. list of connected devices (don't know menu names in your language, mine is in french 😉).
Do you have one or several devices recorded?

If yes remove them, and try pairing a new device.
- most of the time pairing works on first trial, but from time to time you have to try 2 or 3 times.

Philoul, 2025-02-22:
> in our case we needed to install the mysugr app and pair via that app first. We keep the BT connection via mysugr app and then pair in AndroidAPS.

I don't understand, AFAIK you cannot have several app connected to your pump at the same time but only one...

Can you remove pairing with mysugr app and then try again pairing with AAPS and verify if it works correctly ?

kneuZ, 2025-02-22:
> Within pump UI If you go to device settings, communication. list of connected devices (don't know menu names in your language, mine is in french 😉). Do you have one or several devices recorded?
>
> If [...]

I have deleted all connected devices in my pump. 
I have tried very often to pair the devices, unfortunately without success. 

I have also just tried with the mysugr app, also without success. 

So I have deleted the connection in my cell phone and in the pump, I get the pump displayed for pairing but no connection is established. 
In the old version it works without any problems. 

When I update the AAPS to version 3.3.1.3 with a paired pump, I only get a time out under the new version.

Philoul, 2025-02-22:
> When I update the AAPS to version 3.3.1.3 with a paired pump, I only get a time out under the new version.

Very strange, update from 3.2 to 3.3 should keep the settings and the pairing...
So an Insight pump paired and working with AAPS 3.2 should continue to work directly after the update of AAPS in 3.3 (without pairing again your pump).

Do you have another phone to test if pairing issue is the same?

Another possible test is to temporary downgrade AAPS to 3.2, pair again pump with AAPS and then upgrade to 3.3 (you will loose database, but an NSClient full sync should recover it) 🤔

@MilosKozak @TebbeUbben Do you have any ideas on how to track this issue?
Unfortunatly I don't have the skills to analyse BT connection or analyse pairing process (when I managed the migration of Insight driver to kotlin, I made hundred of verification to keep communication and pairing working the same way).

Philoul, 2025-02-22:
Just managed quickly on my dev conf a new pairing, in less than 20s Î saw the pump, selected it, approved the key on both sides and connection was ok.

KelvinKramp, 2025-02-22:
@Philoul your right also works without mysugrapp. using poxel6a here.

Philoul, 2025-02-22:
my sugr app not installed (in none of my phones, loop or dev)

reijoahola-nightscout, 2025-02-23:
I remember that A-C Insight has always supported two BT connections at the same time. For example one between pump and A-C remote controller, one between pump and Android phone. I have not tested with mySugr app.

reijoahola-nightscout, 2025-02-23:
Insigh version in this issue seems to be 3.0. In my Insight the version is 2.01. This could cause some differences in BT pairing behaviour.

reijoahola-nightscout, 2025-02-23:
I have now used AAPS with Insight over 4 years. I have always had some problems with BT pairing, typically it has never paired with the first try and several times  is needed. A couple of times it did not succeed without special operations. The document mentions that when problems with pairing "... turn off bluetooth on pump AND smartphone for about 10 seconds and then turn it back on.". This has always solved the problem.

Philoul, 2025-02-23:
> Insigh version in this issue seems to be 3.0. In my Insight the version is 2.01. This could cause some differences in BT pairing behaviour.

No, pairing works with both versions, v2 and v3.

> I have always had some problems with BT pairing, typically it has never paired with the first try and several times is needed.

I agree that pairing process is the most tricky part within driver. On my side I never got so bad results but efficiency of pairing process can be linked to BT implementation within phone.
During driver migration in kotlin hundred of pairing has been done but only with my dev and loop phone: Android 11 for both phones (Samsung A10 and Power U20), Insight pump v2 and v3.
About 90% pairing worked the 1rst time, and 10% It succeed on second trial.

> ... turn off bluetooth on pump AND smartphone for about 10 seconds and then turn it back on.". This has always solved the problem.

👍Good idea, on pump this can be done from menu, or just by removing battery

kneuZ, 2025-02-26:
Today I tried again.
The pump is coupled with version 3.2.0.2. I update to Verision 3.3.1.3, the pump is still paired, but no connection is established (log attached). 

[image]

[AndroidAPS(After Update).log](https://github.com/user-attachments/files/18985770/AndroidAPS.After.Update.log)

I decoupled the pump, deleted the connection in the pump and switched off the cell phone and pump for over 10 seconds. 
As soon as I try to pair the pump again, the message “Connection failed” appears immediately.
I have also made logs of this. 

[AndroidAPS(New Paring).log](https://github.com/user-attachments/files/18985789/AndroidAPS.New.Paring.log)

After the “downgrade” to version 3.2.0.2, the pairing works again immediately and without any problems.

Unfortunately, I don't have another smartphone to test. I have been using AAPS with the Insight for several years, and unfortunately the problems only started with this version.

Philoul, 2025-02-26:
When I look within log, I find this:
```
15:22:14.372   [RecoveryTimer] I/PUMP: [InsightConnectionService.setState():244]: Insight   state changed: CONNECTING
--
15:22:14.379 [Thread-24] I/PUMP:   [InsightConnectionService.handleException():327]: Exception occurred:   ConnectionFailedException
15:22:14.380 [Thread-24] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: RECOVERING
15:22:22.400 [RecoveryTimer] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: CONNECTING
15:22:22.412 [Thread-26] I/PUMP:   [InsightConnectionService.handleException():327]: Exception occurred:   ConnectionFailedException
15:22:22.419 [Thread-26] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: RECOVERING
15:22:26.474 [main] I/PUMP:   [InsightConnectionService.withdrawConnectionRequest():272]: Last connection   lock released, will disconnect in 5 seconds
15:22:26.479 [main] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: NOT_PAIRED
15:24:51.228 [DefaultDispatcher-worker-1]   D/PUMP: [VirtualPumpPlugin.refreshConfiguration():405]: Pump in   configuration: Accu-Chek Combo, PumpType object: ACCU_CHEK_COMBO
15:24:51.230 [DefaultDispatcher-worker-1]   D/PUMP: [VirtualPumpPlugin.refreshConfiguration():407]: New pump   configuration found (ACCU_CHEK_COMBO), changing from previous (null)
15:26:01.787 [main] I/PUMP:   [InsightConnectionService.pair():379]: Pairing initiated
15:26:01.805 [main] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: CONNECTING
15:26:01.823 [Thread-17] I/PUMP:   [InsightConnectionService.handleException():327]: Exception occurred:   ConnectionFailedException
15:26:01.827 [Thread-17] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: NOT_PAIRED
15:26:04.941 [main] I/PUMP:   [InsightConnectionService.pair():379]: Pairing initiated
15:26:04.952 [main] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: CONNECTING
15:26:04.962 [Thread-18] I/PUMP:   [InsightConnectionService.handleException():327]: Exception occurred:   ConnectionFailedException
15:26:04.977 [Thread-18] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: NOT_PAIRED
15:26:09.647 [main] I/PUMP:   [InsightConnectionService.pair():379]: Pairing initiated
15:26:09.662 [main] I/PUMP:   [InsightConnectionService.setState():244]: Insight state changed: CONNECTING
```
It's the first time I see this kind of error, but everything seems to be within `InsightConnectionService.kt` file...
My 2 different ideas are:
- maybe increase some timeout during connection???
- Revert this file from InsightConnectionService.kt to InsightConnectionService.java (but complicated because a lot of refactoring has been done between 3.2 and 3.3)

@TebbeUbben during driver migration to kotlin, I was not able to convert `ConnectionEstablisher.java` file to kt (it's the only file remaining in java...)
- communication worked with kt version (see `ConnectionEstablisherXX.kt` file within kotlin+java folder), but it was impossible to manage pairing with kt version...

Do you think we can adjust something to make the driver more "tolerant"? to allow pairing with ConnectionEstablisher.kt ? or just fix this issue with the java version of ConnectionEstablisher combine with kotlin for the other files of this driver like InsightConnectionService ?
`BluetoothAdapterExtensionKt.safeEnable(bluetoothAdapter, 2000, null);` increase 2000 to wait more ?
increase TIMEOUT_DURING_HANDSHAKE_NOTIFICATION_THRESHOLD or RESPONSE_TIMEOUT in InsightConnectionService ?
other ?
I'm a bit lost with my poor knowledge of BLuetooth management...

TebbeUbben, 2025-02-27:
@Philoul The first step in Insight pairing is always low-level Bluetooth pairing. If the pump is already listed as a paired device in the Android system settings, you must first remove it before establishing a new connection, so that a new pairing is established upon connection. The system cannot reuse an existing pairing. However, this also means that any issues at this stage are likely deeper within the system's Bluetooth stack and beyond the driver's scope.

In order to ensure that the system initiates a pairing, the previous bond is deleted using the following code in [`ConnectionEstablisher.java`](https://github.com/nightscout/AndroidAPS/blob/master/pump/insight/src/main/java/app/aaps/pump/insight/utils/ConnectionEstablisher.java):

```java
if (forPairing && bluetoothDevice.getBondState() != BluetoothDevice.BOND_NONE) {
    try {
        Method removeBond = bluetoothDevice.getClass().getMethod("removeBond", (Class[]) null);
        removeBond.invoke(bluetoothDevice, (Object[]) null);
    } catch (ReflectiveOperationException e) {
        if (!isInterrupted()) callback.onConnectionFail(e, 0);
        return;
    }
}
```

Unfortunately, [removeBond](https://android.googlesource.com/platform/packages/modules/Bluetooth/+/refs/heads/main/framework/java/android/bluetooth/BluetoothDevice.java#2084) is a hidden system API, so we can't invoke it directly using the Android SDK. However, as can be seen, using some reflection hacks, we can still call it.

Please pay very careful attention to `(Class[]) null` and `(Object[]) null`. [`Class::getMethod`](https://docs.oracle.com/javase/8/docs/api/java/lang/Class.html#getMethod-java.lang.String-java.lang.Class...-) and [`Method::invoke`](https://docs.oracle.com/javase/8/docs/api/java/lang/reflect/Method.html#invoke-java.lang.Object-java.lang.Object...-) use vararg parameters, i.e. they allow you to provide an arbitary amount of parameters of a certain type, which are converted to an array by the compiler. You can also pass an array directly and not make use of the vararg nature of the method at all, which is what the code above does. Casting `null` ensures `null` is passed as an array rather than an array containing `null`.

Now, let's look at your Kotlin code:
```kotlin
if (forPairing && bluetoothDevice.bondState != BluetoothDevice.BOND_NONE) {
    try {
        val removeBond = bluetoothDevice.javaClass.getMethod("removeBond", null)
        removeBond.invoke(bluetoothDevice, null)
    } catch (e: ReflectiveOperationException) {
        if (!isInterrupted) callback.onConnectionFail(e, 0)
        return
    }
}
```

This code will do the latter and create an array containing `null`.  My assumption is that this leads to the reflective operation failing and thus not unpairing the pump. In Kotlin, you have to explicitly instruct the compiler to use a provided array for varargs using the [spread operator](https://kotlinlang.org/docs/functions.html#variable-number-of-arguments-varargs) (`*`). You can also try to just remove the `null` parameter at all and see if that works. Alternatives might be `*emptyArray()` or `*null`, `*(null as Array<Class>?)`, `*(null as Array<Any>?)` to stay closer to the Java version. However, I haven't tested and verified this. You can also try to manually delete the pump in the system settings to see if the issue really boils down to `removeBond` not being called.

Philoul, 2025-03-02:
@TebbeUbben Thanks for the hints and explainations, this gives me solutions to migrate the latest java file.
But I'm afraid it will not fix current issue:
- here an Insight pump correctly paired with AAPS 3.2 stop to connect with AAPS after 3.3 update 🤔 (with ConnectionFailedException)
Never got this issue despite hundred of tests done, so I can't reproduce it.

I think this `ConnectionFailedException` prevent communication (with a pump correctly paired previously) or for new pairing trials (same `ConnectionFailedException` error)

Do you know if we can adjust some parameters or values to make connection more "tolerant" (increase timeout? other?)

TebbeUbben, 2025-03-04:
@Philoul ConnectionFailedException is a very general exception. It just indicates that the phone was not able to establish a connection on the Bluetooth/link layer. The connection fails before the driver has any means to do something. I can't think of anything that might cause this issue going from 3.2 to 3.3 since the code responsible for the Bluetooth connection remained unchanged.

Possible debugging endpoints might be:
- [This line](https://github.com/nightscout/AndroidAPS/blob/master/pump/insight/src/main/kotlin/app/aaps/pump/insight/connection_service/InsightConnectionService.kt#L703) and closer inspect the original exception thrown by the Android system to see if it provides a more detailed error description.
- Inspect the Android system logs with regards to the Bluetooth stack.

Philoul, 2025-03-09:
@kneuZ can you please test my PR #3875 and tell me if you can connect with your Insight pump with this version ?

kneuZ, 2025-06-27:
Hello,

I apologize for my late response. I had been looping with the Medtrum pump in the meantime and had completely forgotten about the Insight problem. 
Today I switched back to the Insight due to allergic reactions. Unfortunately, it doesn't work with version 3.3.2 either. I get the pump displayed for pairing, but the app crashes as soon as I select the pump. Unfortunately, sending a log does not work either. Nothing happens when I press the button.

Is there any other idea what I can try?

[image]

[image]

Philoul, 2025-06-27:
Just after the crash few minutes max, can you connect your phone to a computer an open file browser to manually find log files
//Phone/Android/data/info.nightscout/androidaps/files/*.zip
Take the 2 most recent zip files starting by AndroidAPS.2025... and the file AndroidAPS.log file too

kneuZ, 2025-06-27:
I have the folder \android\Android\data\info.nightscout.androidaps\files

but it is empty even after the crash :-(

kneuZ, 2025-07-01:
Today I tried again to connect my Insight in version 3.2.0.4. This worked without any problems. I then updated to version 3.3.2.0 and the insulin pump remained connected. As soon as I try to connect to the pump, AAPS crashes and shuts down completely.

[AndroidAPS.log](https://github.com/user-attachments/files/20998118/AndroidAPS.log)

closed 2026-10-11 by MilosKozak