nightscout/AndroidAPS #3852: Request for Direct Connection Module for Linx GCM

pjorquerad, 2025-02-26:
Hi there,

First of all, thank you so much for the incredible work you do. It's truly amazing and has made our lives so much easier.

I have a question: why does AAPS include a module for OTTAI GCM, which can't be marketed in many regions, while Linx from Microtech, which has CE certification, doesn't have a module for direct connection? Adding support for Linx would be incredibly helpful.

In Chile, we don't have a wide variety of sensors, and the Libre2+ hasn't been a seamless replacement for the FreeStyle Libre 1. I would greatly appreciate it if someone could add direct connection support for Linx, which is a fairly acceptable alternative in terms of price, quality, and convenience here in Chile and other LATAM countries.

I really appreciate your support and dedication.

Best regards,
PJ

MilosKozak, 2025-02-26:
broadcast support depends on manufacturer.
Contact your reseller to push on Microtech

pjorquerad, 2025-02-26:
Hi Milos,

Thank you for your response and please forgive my ignorance in these matters. I'm just a user of AAPS for my young son (7 years old) and not a programmer. I initially thought it was the other way around—that you, with your superpowers, adjusted the AAPS code to make it compatible with the sensor. Now I understand that it depends on the manufacturer.

I am currently speaking with the distributor in Chile who has direct contact with Microtech. Could you please indicate specifically what you require from them? Is it a firmware update that is compatible with AAPS? Documentation? The contact will likely be commercial and may not fully understand what is needed. The more information we can provide, the better.

I'm surprised they have a module for the Equil 5.3 pump but haven't provided the material for Linx GCM. Let's see what happens!

Thank you so much for your great support as always!

pjorquerad, 2025-02-26:
Hi Andy,

Thank you very much for your intervention. I thought that what you suggested might work, but unfortunately, I couldn't get Xdrip to directly receive transmissions from the native Linx App when selecting Aidex as the data source in Xdrip. (I imagine that the previous Aidex sensor, being larger and having a transmitter base, must be quite different from the Linx GCM, which is very similar to a Libre 3.)

As a result, the only thing that has worked is to leave Xdrip in Companion App mode so that it reads notifications from the Linx app and extracts glucose readings from there. (This method is indicated in the Xdrip documentation, and Linx App is listed as compatible as of 17.12.2024.) I am testing it on myself, and I don't have type 1 diabetes, and it works quite well. However, sometimes, without any apparent reason, my phone's software stops showing notifications from the Linx app, even though I have given it all possible permissions. It's strange; I can go a whole day with the Linx app closed and still receive notifications, but on some days, it stops despite having permission to run in the background and without battery restrictions, etc. I've given it everything I can... but still, notifications sometimes disappear.

That's why I'm interested in finding a cleaner solution that doesn't cause interruptions and is safe for my son's treatment.

Thank you very much for your support!

Best regards,

MilosKozak, 2025-02-26:
Aidex was removed because there was zero users for a longer time.
Anyway I know people at Microtech and they know me. They can contact me anytime. But the push must be from someone else (you)

andyrozman, 2025-02-26:
I am afraid you won't find any cleaner solution. When I was adding Aidex to Xdrip I was looking in different ways this could work, I even had my own version of Aidex app, but app was relatively big, so adding "brains" into Xdrip was not an option (to cut out their official app), because that would have to also be maintained... So adding receiving of broadcasts was next best solution, since people at Microtech were very eager to make their CGMS be supported by wider audience. So I supplied them with code what they need to do, and they added it to their app, and in turn Xdrip support was added (receiving thoose broadcasts), and same support was added to AAPS. 
But problem was the same, if App stopped working for whatever reason broadcast would stop and same is happening to you now. Their sensors didn't allow direct connection to sensors, you always had to discover them (BT Discovery scan), which required that Location (GPS) is always on...

pjorquerad, 2025-02-26:
Wow, what an impressive job you've done, my friend! Thank you so much for all your hard work! It's truly remarkable, really!

So, it sounds like we need to find the perfect device where we can give the native Linx app all possible permissions to ensure it never closes. I'll continue testing, and I'm already in contact with the Linx distributor in Chile to press Microtech and have them get in touch with Milo as well.

Thank you both for your support once again!

MilosKozak, 2026-10-11:
merged in v4

closed 2026-10-11 by MilosKozak