nightscout/AndroidAPS #3866: Manual automation not changing profile %

robertrub, 2025-02-28:
I have an _activity_ manual automation. Sets profile to 30% and TT to 150.

The _set TT to 150_ is working but _set profile to 30%_ or 50% isn't working. 

[image]

[image]

[image]

Even without the TT, the profile % doesn't change.

Here are the logs. Look around 15h50 and later.

[image]

Using last dev

[image]

robertrub, 2025-02-28:
When I try to change profile manually,  I get this strange message and the profile % doesn't change either.

[image]

robertrub, 2025-02-28:
Manually, I cannot change **the profile %** but if I change the **profile AND the profile %**, then it works.

robertrub, 2025-03-01:
Very strange, it's working normally this morning...

olorinmaia, 2025-03-01:
I thought the limit for automations was 50% ;)

I can see a basal rate in your profile is 0,15. So 0,15 * 0,3 = 0,045 which is below the lowest basal rate that can be set on pump at the moment. I believe that is why you get this error message if you use that profile and try to execute automation rule.

robertrub, 2025-03-01:
As I said, I got the same error when changing profile to 50% too... It was something else but what ???

FYI, I changed the automation limits to 30 and 150% for my personal use.

MilosKozak, 2025-03-01:
I testes 50% on latest dev and no issues (virtual pump)
Test unmodified version

MilosKozak, 2025-03-01:
Hmmm no idea what happened. Pump was reporting profile already set correctly but AAPS didn't see it

closed 2025-03-01 by MilosKozak