nightscout/AndroidAPS #3890: Tidepool Integration: Authentication Changes

tjotala, 2025-03-18:
Tidepool is planning to deprecate our legacy authentication APIs in favor of new OAuth2 / OpenID Connect standards based APIs. You can find the documentation here:

https://developer.tidepool.org/TidepoolApi/#tag/Authentication/operation/Authorize
https://developer.tidepool.org/TidepoolApi/#tag/Authentication/operation/ObtainToken

We see that AndroidAPS is currently using the legacy authentication APIs: https://github.com/nightscout/AndroidAPS/blob/master/plugins/sync/src/main/kotlin/app/aaps/plugins/sync/tidepool/comm/TidepoolApiService.kt#L28-L29

Please contact us to discuss migration to the new authentication APIs:

https://developer.tidepool.org/
info@tidepool.org

Sincerely,

Tapani Otala
VP, Engineering
Tidepool

MilosKozak, 2025-03-29:
[image]
@tjotala what is allowed clientId ?
"AAPS" didn't pass

MilosKozak, 2025-03-29:
tha same for redirectUri "aaps://callback/tidepool"

toddkazakov, 2025-04-15:
Hi @MilosKozak, here are the client details:

Integration:
- Authorization Endpoint: https://auth.integration.tidepool.org/realms/integration/protocol/openid-connect/auth
- Token Endpoint: https://auth.integration.tidepool.org/realms/integration/protocol/openid-connect/token
- Well Known: https://auth.integration.tidepool.org/realms/integration/.well-known/openid-configuration
- API Host: https://external.integration.tidepool.org
- Client ID:`aaps`
- Callback URL: `aaps://callback/tidepool`

Production:
- Authorization Endpoint: https://auth.tidepool.org/realms/tidepool/protocol/openid-connect/auth
- Token Endpoint: https://auth.tidepool.org/realms/tidepool/protocol/openid-connect/token"
- Well Known: https://auth.tidepool.org/realms/tidepool/.well-known/openid-configuration
- API Host: https://api.tidepool.org
- Client ID:`aaps`
- Callback URL: `aaps://callback/tidepool`

Let me know if I can assist you in any other way. Thanks!

MilosKozak, 2025-09-25:
@toddkazakov some people have problems displaying data in Tidepool (mostly profile updates).
Is there someone who can help on Tidepool side and analyzi the issue. I did my best and data is accepted by Tidepool